pvlib/pvlib-python #2: bring pvlib-python up to date with PVLIB-MATLAB 1.2

Calama-Consulting, 2015-02-17:
Add additional functions introduces in PVLIB-matlab 1.2

dacoex, 2015-02-17:
Where can we find PVLIB-matlab 1.2

wholmgren, 2015-02-17:
PVLIB-Matlab 1.2 is only available at the PVPMC website. The github PVLIB_Matlab repository is an older version and may be removed. @Calama-Consulting you might want to ask Josh if he wants the matlab project available on github since it may hurt their official download statistics.

The readme has a link to the documentation.

We want to keep the core functionality and algorithms of the python and matlab projects roughly in parallel, but we're not making any promises at this point. 

Yes, please do make separate issues for any feature requests.

dacoex, 2015-02-17:
>PVLIB-Matlab 1.2 is only available at the PVPMC website. The github PVLIB_Matlab repository is 
an older version and may be removed.

OK, so login and find out...

>The readme has a link to the documentation.

Which is behind a login gate.

>We want to keep the core functionality and algorithms of the python and matlab projects roughly in >parallel, but we're not making any promises at this point. 

So this means:
PVPMC adds a functionality and then issues a release, then the pvlib-python would be updated?
A rather parallel process?
In consequence, one would have to know Mtalab, as well?

>Yes, please do make separate issues for any feature requests.

OK, one the development strategy discussed here is clear.

wholmgren, 2015-02-17:
The read me for this github repo has a line

"Full documentation can be found at readthedocs"

Which links to

http://pvlib-python.readthedocs.org/en/latest/

This is the Pvlib Python documentation only.

The Matlab and Python projects are developed by different teams that do not
regularly work together. There will not be feature parity, only similarity.

dacoex, 2015-02-17:
>The Matlab and Python projects are developed by different teams that do not regularly work together. There will not be feature parity, only similarity.

OK, this is more clear.
Consequently, you may decide to spearhead with new features and then the Matlab will follow?

Or is pvlib after a similar feature set but more driven by its developers/users?

Maybe good also to limit the scope and compare it to other packages such as https://github.com/pingswept/pysolar. Then teh docs cloud show where pvlib by its intention will be different apart from the more sphisticated tied-in to the scipy stack with the reliance on pandas (very positive).

I am thinking of something like:
Related Packages
http://statsmodels.sourceforge.net/devel/related.html

Looking forward to your feedback on this and the above questions.

wholmgren, 2015-02-17:
Pvlib Python already has a few features that the Matlab version doesn't
have. It's up to the Matlab developers to update their code where possible.
I can only say that the Python project will try to stay current with most
things that the Matlab project adds.

The Python project will be mostly community supported, whereas the Matlab
project is mostly supported by u.s. department of energy grants.

The Python project is very much tied to the scientific Python stack. This
would not be feasible without those libraries.

Our pvsc 2015 abstract and rob's 2014 PVSC paper outline the scope of this
project.

Pysolar is much more limited in scope and is gpl licensed. Nothing against
Pysolar, it's just different.

I will put some of this in the docs/readme.

dacoex, 2015-02-18:
Thanks a lot.

So next could be a series of feature plans that can be added to the roadmap.

wholmgren, 2015-04-09:
Here's a list of the v1.2 Matlab project's functions that we're missing (from the 1.2 help page):

- [ ] pvl_louche - Estimate Direct Normal Irradiance from Global Horizontal Irradiance using the LOUCHE model
- [ ] pvl_orgill_hollands - Estimate Direct Normal Irradiance from Global Horizontal Irradiance using the Orgill and Hollands model
- [ ] pvl_erbs - Estimate Direct Normal Irradiance from Global Horizontal Irradiance using the Erbs model
- [ ] pvl_reindl_1 - Estimate Direct Normal Irradiance from Global Horizontal Irradiance using the Reindl 1 model
- [ ] pvl_reindl_2 - Estimate Direct Normal Irradiance from Global Horizontal Irradiance using the Reindl 2 model
- [ ] pvl_martinruiziam - Determine the incidence angle modifier using the Martin and Ruiz incident angle model
- [ ] pvl_adrinverter - Anton Driesse's Grid-Connected PV Inverter

I also looked at the changelog but most of it has either already been incorporated or is Matlab specific.

Many of these would be good first PRs.

bmu, 2015-04-10:
Maybe Anton could implement "his Inverter" ;-) @adriesse?

adriesse, 2015-04-10:
Sure.  What's the best way to publish a
      collection of model parameters?
      
      Anton
      
      
      On 2015-04-10 8:05, bmu wrote:
    
    
      Maybe Anton could implement "his Inverter" ;-) @adriesse? 
      —
          it on GitHub.

wholmgren, 2015-04-10:
Thanks. I suggest a csv file in ``pvlib/data`` since this can be easily imported by pandas.

wholmgren, 2016-01-12:
Copy/paste from PVPMC's recent PVLIB Matlab 1.3 announcement...

- [ ] pvl_FSspeccorr: Spectral mismatch modifier function contributed by First Solar based on precipitable water.
- [ ] pvl_calcPwat: function to estimate precipitable water content
- [ ] pvl_huld: PV performance model of Huld et al., 2011
- [ ] pvl_PVsyst_parameter_estimation: function to estimate PVsyst module parameters from IV curves.
- [ ] pvl_calcparams_PVsyst: Calculates the five parameters for an IV curve using the PVsyst model.
- [ ] pvl_desoto_parameter_estimation: function to estimate Desoto module parameters from IV curves.
- [ ] pvl_getISDdata: Functions to access ground measured weather data from NOAA's Integrated Surface Data network

cwhanse, 2016-01-12:
Will,

I put version 1.3 together.  The PVPMC notice is only highlights, here’s the full list of new functions and changes copied from the Matlab toolbox release notes.

Cliff

% *New Functions*
%
% |pvl_PVsyst_parameter_estimation| - Estimates parameters for the PVsyst
% version 6 module performance model.
%
% |pvl_desoto_parameter_estimation| - Estimates parameters for the De Soto
% single diode module performance model.
%
% |pvl_rectify_IV_curve| - Ensures that Isc and Voc are included
% in a IV curve and removes duplicate voltage and current points.
%
% |pvl_calcparams_CEC| - Create module performance model coefficients
% for the single diode model used by the CEC.
%
% |pvl_calcparams_PVsyst| - Create module performance model coefficient
% for the single diode model in PVsyst version 6.
%
% |pvl_huld| - Calculates DC power using the Huld PV module model.
%
% |pvl_calcPwat| - Calculate precipitable water (cm) from ambient
% air temperature (C) and relatively humidity (%).
%
% |pvl_FSspeccorr| - Calculate spectral mismatch modifier based on
% precipitable water and absolute airmass.
%
% |pvl_getISDdata| - Fetch data from NOAA ISD at ftp.ncdc.noaa.gov.
%
% |pvl_readISH| - Read data fetched from ftp.ncdc.noaa.gov into a table.
%
% |pvl_lambertw| - Compute values for the Lambert W function W(z).
%
% |numdiff| - Compute numerical derivatives for unequally spaced data.
%
% *Significant Changes*
%
% # Corrected |pvl_erbs|; the coefficient of K_t^2 changed from 4.3888 to
% 4.388
% # Changed |pvl_calcparams_desoto| to require effective irradiance as its
% input, rather than absorbed irradiance (i.e., broadband irradiance
% reaching a module's cells). |pvl_calcparams_desoto| no longer computes and
% applies a spectral mismatch modifier.
% # Changed |pvl_calcparams_desoto| to remove the requirement that all inputs
% be vectors of the same length.  Inputs |S| and |Tcell| can be vectors of the
% same length; all other inputs should be scalars.
% # Replaced |wapr_vec| with |pvl_lambertw|. |wapr_vec| is obsolete and removed.
% # Changed |pvl_haydavies1980| and |pvl_perez| to accept |NaN| in inputs.
% # Added functions |pvl_FSspeccorr| and |pvl_calcPwat| to implement
% FirstSolar’s spectral mismatch modifier model.
% # Removed |pvl_snlinverterdb| (made obsolete by
% |pvl_SAMLibraryReader_SNLInverters| and |SandiaInverterDatabaseSAM2014.1.14.mat|).
% # Changed |pvl_erbs|, |pvl_louche|, |pvl_orgill_hollands|, |pvl_reindl_1|,
% and |pvl_reindl_2| to output DNI, DHI and Kt, and to clarify each model’s description.
% # Corrected description of |pvl_kingdiffuse| to clarify that the King
% diffuse model estimates total plane-of-array diffuse (sky diffuse and ground reflected irradiance).
% # Included databases of CEC module model coefficients (|CECModuleDatabaseSAM2015.6.30.mat|) and Sandia inverter
% model coefficients (|CECInverterDatabaseSAM2015.6.30.mat\) from SAM2015.6.30.

wholmgren, 2016-01-12:
Thanks Cliff!

wholmgren, 2016-03-18:
In anticipation of the upcoming Santa Clara workshop, I'm consolidating the lists in the comments above into a new list. I also removed some of the items that don't apply to python. It's probably best to start making new issues for specific features.

- [ ] pvl_louche - Estimate Direct Normal Irradiance from Global Horizontal Irradiance using the LOUCHE model
- [ ] pvl_orgill_hollands - Estimate Direct Normal Irradiance from Global Horizontal Irradiance using the Orgill and Hollands model
- [ ] pvl_erbs - Estimate Direct Normal Irradiance from Global Horizontal Irradiance using the Erbs model. And from v1.3: Corrected pvl_erbs; the coefficient of K_t^2 changed from 4.3888 to 4.388
- [ ] pvl_reindl_1 - Estimate Direct Normal Irradiance from Global Horizontal Irradiance using the Reindl 1 model
- [ ] pvl_reindl_2 - Estimate Direct Normal Irradiance from Global Horizontal Irradiance using the Reindl 2 model
- [ ] pvl_martinruiziam - Determine the incidence angle modifier using the Martin and Ruiz incident angle model
- [ ] pvl_adrinverter - Anton Driesse's Grid-Connected PV Inverter
- [ ] pvl_PVsyst_parameter_estimation - Estimates parameters for the PVsyst
  version 6 module performance model.
- [ ] pvl_desoto_parameter_estimation - Estimates parameters for the De Soto
  single diode module performance model.
- [ ] pvl_rectify_IV_curve - Ensures that Isc and Voc are included
  in a IV curve and removes duplicate voltage and current points.
- [ ] pvl_calcparams_CEC - Create module performance model coefficients
  for the single diode model used by the CEC.
- [ ] pvl_calcparams_PVsyst - Create module performance model coefficient
  for the single diode model in PVsyst version 6.
- [ ] pvl_huld - Calculates DC power using the Huld PV module model.
- [ ] pvl_calcPwat - Calculate precipitable water (cm) from ambient
  air temperature (C) and relatively humidity (%). see #115.
- [ ] pvl_FSspeccorr - Calculate spectral mismatch modifier based on
  precipitable water and absolute airmass. see #115.
- [ ] pvl_getISDdata - Fetch data from NOAA ISD at ftp.ncdc.noaa.gov.
- [ ] pvl_readISH - Read data fetched from ftp.ncdc.noaa.gov into a table.
- [ ] Changed pvl_calcparams_desoto to require effective irradiance as its
input, rather than absorbed irradiance (i.e., broadband irradiance
reaching a module's cells). pvl_calcparams_desoto no longer computes and
applies a spectral mismatch modifier.
- [ ] Changed pvl_calcparams_desoto to remove the requirement that all inputs
be vectors of the same length.  Inputs S and Tcell can be vectors of the
same length; all other inputs should be scalars.
- [ ] Changed pvl_erbs, pvl_louche, pvl_orgill_hollands, pvl_reindl_1, and pvl_reindl_2 to output DNI, DHI and Kt, and to clarify each model’s description.
- [ ] Corrected description of pvl_kingdiffuse to clarify that the King diffuse model estimates total plane-of-array diffuse (sky diffuse and ground reflected irradiance).
- [ ] Added CEC module library and inverter library from SAM2015.6.30 in .csv format and in .mat format, along with scripts to read the .csv files and write the .mat files.

jforbess, 2016-03-18:
I intend to create a function to calculate the sandia inverter coefficients
from datasheet and CEC efficiency information. I was planning to use Santa
Clara as my motivation to get it done, either before or during.

I think it would be considered another in the parameter_estimation set of
functions?

cwhanse, 2016-03-19:
Absolutely. In fact I have some matlab code already but it's not in pvlib format yet.

-

wholmgren, 2016-03-19:
@jforbess that sounds great. I suggest creating a new issue if you'd like to discuss it further.

jsstein, 2016-03-21:
Just wanted to let folks know that the Matlab help files are available at a new location on the PVPMC website:  [https://pvpmc.sandia.gov/PVLIB_Matlab_Help/](url)

I have been updating and cleaning up the website and several old links no longer work.  Please let me know if you find any problems.

wholmgren, 2016-03-21:
Thanks @jsstein, I'll add a link to this page from our documentation.

adriesse, 2023-04-03:
I suggest we close this issue and open more specific ones when needed.

cwhanse, 2023-04-03:
Of the remaining items, Huld's DC model and perhaps the read ISD/ISH functions are worth considering. I wouldn't object to adding the Orgill and Hollands model. I would consider removing `pvlib.irradiance.king` since there will never be an archival publication to reference.

mikofski, 2023-04-03:
PVGIS is using Huld model. What data are in ISD/ISH? Is it just rain & snow data? How are these data archived? API, FTP, or HTTP? Maybe this could be part of `iotools`? What about GSOD data?

cwhanse, 2023-04-03:
ISD is global data primarily for aviation. Sometimes (not common) there is an irradiance sensor, but mostly I've used it for air temperature and wind. https://www.ncei.noaa.gov/products/land-based-station/integrated-surface-database

wholmgren, 2023-04-03:
Issues for ISD and GSOD: #759 #668 

I'm going to close this but other maintainers can reopen it if they feel it still has value.

closed 2023-04-03 by wholmgren

cwhanse, 2023-04-03:
I feel that we should have some kind of distributed celebration for closing _the_ oldest issue.