pvlib/pvlib-python #4: fix dirint function

Calama-Consulting, 2015-02-17:
similar to Sandia-Labs/PVLIB_Python#40 , ensure compatibility in the new package

wholmgren, 2015-03-12:
Wondering if this is going to make it into 0.1 and if so, who is going to do it. I would consider implementing it for 0.2, but I can't do it in the 0.1 time frame.

Calama-Consulting, 2015-03-12:
@wholmgren , is this completed with #26 ?

wholmgren, 2015-03-12:
No, that PR just addressed the issues with the DISC function.

Calama-Consulting, 2015-03-12:
Right, lets shift this to 0.2 for the dirint function.

wholmgren, 2015-03-24:
PVLIB MATLAB 1.2 has a lot more GHI to DNI functions: 
```
* pvl_louche
* pvl_orgill_hollands
* pvl_erbs
* pvl_reindl_1
* pvl_reindl_2
```
Not sure if we want to rename this issue to encompass the implementation of all of these algorithms or if we should just put this in #2.

PR #44 (2015-04-04): implement dirint
This PR implements the dirint function based on the MATLAB ``pvl_dirint.m`` file. Closes #4.

I've not yet tested the implementation very thoroughly, so let's hold off on merging until I write a few tests.

As suggested in #42, I would prefer to put this function and the disc function in ``irradiance.py`` instead of ``clearsky.py``.
closed 2015-04-04 by bmu