pvlib/pvlib-python #6: Missing values using DISC to generate the direct normal irradiance

Lamaf, 2015-02-25:
I am having missing values when using DISC to simulate DNI, I trad this using clear sky GHI data and even with data I have from a measuring device 
For an example try this:

    import datetime
    import pandas as pd
    from pvlib.location import Location
    import pvlib.solarposition
    import pvlib.clearsky
    
    # make a location
    tus = Location(32.2, -111, 'MST', 700)
    # make a pandas DatetimeIndex for some day
    times = pd.date_range(start=datetime.datetime(2014,6,24), end=datetime.datetime(2014,6,25), freq='1Min')
    # calculate the solar position
    solpos = pvlib.solarposition.get_solarposition(times, tus, method='pyephem')
    # calculate clear sky data
    tus_cs = pvlib.clearsky.ineichen(times, tus, airmass_model='young1994')
    ax = tus_cs.plot()
    # calculate the direct normal irradiance using GHI from the clear sky data
    DNI= pvlib.clearsky.disc(Time= solpos.index, GHI=tus_cs.GHI, SunZen=solpos.zenith)
    DNI.DNI_gen_DISC.plot(ax=ax, legend=True)

Am I doing something wrong ??

[image]

wholmgren, 2015-02-25:
Thanks for moving the issue here.

Looks like there are some nan assignments:

https://github.com/pvlib/pvlib-python/blob/master/pvlib/clearsky.py#L378

I don't personally have any experience with this algorithm so I don't know if those conditions can be relaxed or not. If you have experience with this algorithm then I'll encourage you to make the appropriate changes and submit a pull request. The documentation needs improvement too.

Lamaf, 2015-02-26:
It is not the nan assignments in the script. According to the DISC algorithm, the condition in the lines 368,369 and 370 must be (Kt <= 0.6)
Still not sure if it perfect, as in my previous example the values of DNI is high in the early morning and in the end of the day, because  the GHI will be more than the I0h
and here is the plot
[image]
However in the measured data till now it's sufficient result

wholmgren, 2015-02-26:
Ok, I understand now. 

Sandia-Labs/PVLIB_Python#35 fixed the DISC algorithm on the old master branch.

Sandia-Labs/PVLIB_Python#37 was supposedly closed with [Sandia-Labs/PVLIB_Python/8ffbf80e](https://github.com/Sandia-Labs/PVLIB_Python/commit/8ffbf80e6935294f9b165175174047d9f184b50c), but I now see that the fix was incomplete since it missed the ``0.6-1``'s on the right side of the assignments.

Better. Unit. Tests.

@Lamaf or @Jorissup, do either of you want to make a pull request to fix this? You'd also need to fix the documentation and add a unit test to [test_clearsky.py](https://github.com/pvlib/pvlib-python/blob/master/pvlib/test/test_clearsky.py). I would also like to see some PEP8 applied to this algorithm, but that is less important.

commit 14d9a0f7 (2015-03-12): clean up and fixes. closes #6, closes #24, closes #25
closed 2015-03-12 by wholmgren