pvlib/pvlib-python #11: Versioning and labels

bmu, 2015-03-01:
I think we should use github milestones. Also it is not clear to me on which version we are at the moment (`pvlib.version` says 0.3, but we are on something like `0.4dev` I think).

I would also like to use github labels for the issues.

wholmgren, 2015-03-02:
Agreed on both. I was wishing for milestone labels yesterday. 

@Calama-Consulting did you have a plan for what version to call the first PyPI release? 1.0 would make some sense for a first release, but I also feel like there's still a lot of loose ends for that label. So, I have a slight preference for 0.4.

Calama-Consulting, 2015-03-03:
I agree with @wholmgren , the initial release will probably not warrant the 1.0 label, but I'd like to start at a fresh versioning with the new package. Let's reset  and release initially under 0.1? If there's agreement I'll set up the tags appropriately

wholmgren, 2015-03-03:
Sure, I'm ok with calling it 0.1. One slight hiccup is that anybody that ``pip`` installed the package will need to specify ``-force`` instead of ``-upgrade``. That's probably only a concern for 3 people though.

bmu, 2015-03-12:
I created milestones and some labels. Maybe you should have a look.
If they seem reasonable, we can close this issue.

wholmgren, 2015-03-12:
Nice, they look good.

Let's leave this issue open until we fix the version (it's currently 0.3, which is a hold-over from the UARENForecasting project)

wholmgren, 2015-03-13:
I changed the version to ``0.1.0`` in a28e45d2. I suggest that we keep it this way until after the release, at which point we change it to ``0.1.1dev``. I think that there are some scripts for fancy versioning, but I don't think we need them now. Not sure if you want to close this issue or not.

closed 2015-03-13 by bmu

dacoex, 2015-03-25:
What about adding a label such as "Good as first PR" / "easy catch"

https://github.com/pydata/pandas/labels/Good%20as%20first%20PR

For contibutors with litlle time or who are just starting to understand this library or do not use it everyday.