pvlib/pvlib-python #12: Class based testing

bmu, 2015-03-01:
I would like to implement real nose test cases or pandas based [`TestCases`](https://github.com/pydata/pandas/blob/master/pandas/util/testing.py#L80)

Also pandas like testscripts for lokal testing would be nice (see https://github.com/pydata/pandas).

wholmgren, 2015-03-02:
``nosetests`` will try to run any python modules labeled ``*test*`` that are in the current directory or any subdirectories. You can also give it an explicit test module e.g. ``nosetests pvlib/tests/test_tmy.py``. That being said, I don't have any objection to adding a ``test.sh`` similar to the one in ``pandas``.

There are certainly a lot of tests that could be made cleaner with class based testing and setUp methods.

bmu, 2015-03-02:
Yes, it should, however if I run it in my local repository from the base folder (where `setup.py` is located) the output of `nosetests` is  

```
----------------------------------------------------------------------
Ran 0 tests in 0.000s

OK
```

Not sure whats going on here, as the coverage seems to work:

```
nosetests -v -w pvlib --with-coverage --cover-package=pvlib

Name                  Stmts   Miss  Cover   Missing
---------------------------------------------------
pvlib                    11      0   100%   
pvlib.atmosphere         38     29    24%   56-57, 97-99, 144-146, 215-242
pvlib.clearsky          106     91    14%   106-216, 265-273, 280-283, 350-388
pvlib.irradiance        196    169    14%   85-115, 132, 149, 168, 195-202, 230-238, 267-279, 287-290, 333-363, 428-444, 502-516, 582-586, 677-695, 780-799, 902-926, 983-988, 1115-1186, 1245-1354
pvlib.location           22     15    32%   50-66, 71-73
pvlib.pvsystem          201    177    12%   10-11, 89-105, 161-166, 239-254, 438-458, 521-548, 553-561, 627-660, 725-748, 847-896, 938-969, 978-979, 987-988, 999-1010, 1084-1101
pvlib.solarposition     174    159     9%   41-52, 86-118, 122-133, 160-204, 268-380, 422-440, 456-466
pvlib.spa_c_files         0      0   100%   
pvlib.tmy               106     88    17%   159-182, 187-190, 197-201, 207-213, 218-233, 392-404, 426-438, 443-498
pvlib.tools              46     30    35%   35-36, 57-58, 79-80, 102-103, 120-137, 155-163, 183-186
pvlib.version             1      0   100%   
---------------------------------------------------
TOTAL                   901    758    16%   
----------------------------------------------------------------------
Ran 0 tests in 0.001s

OK
```

wholmgren, 2015-03-02:
Strange. I ran that exact command from the same directory and it worked as expected.

```
$ nosetests --version
nosetests version 1.3.4

$ coverage --version
Coverage.py, version 3.7.1.  http://nedbatchelder.com/code/coverage
```

Calama-Consulting, 2015-03-03:
In some cases you need to enable the --exe option for it to work . That being said the following worked for me:

    nosetests -v 

in the pvlib-python directory

bmu, 2015-03-04:
I use the same versions. However I'm on linux 64bit, maybe something goes wrong here. Still couldn't figure out, what the problem is, but will have time on Friday to check this.

dacoex, 2015-03-04:
@bmu 
take a look at this:
https://launchpad.net/~pythonxy/+archive/ubuntu/pythonxy-devel
may ease up your updating with auto built daily packages.

bmu, 2015-03-08:
I solved my problem (The reason was, that I was trying to test pvlib on a NTFS drive on my new machine, which was mounted so that the files were executable and nosetests skippes files, that are executable!). 

@wholmgren: Do you think it makes sence, that I open a pull request that changes the unit tests to "real" unittests test now, or is this something for a later release? I could also create the milestones (see #11) first.

wholmgren, 2015-03-08:
There are definitely problems with the existing test modules that classes traditionally fix: 
* The set up code is run once at the beginning of the module and reused for all tests 
* The set up code is executed when the module is imported since it's not protected by a class or ``if __name__ == '__main__'`` statement
* I used copy/paste/modify to write many functions

I'll take most (all?) of the blame for the fast and easy but unsustainable pattern. Some of the tests, particularly those that use solar position, will probably need to be made faster if the set up is executed multiple times.

Do you want to remove the nose dependency for development or just clean up the copy/paste mess? 

I'm reluctant to support a rule along the lines of "all tests must be in classes that inherit from pandas ``TestCase`` or ``unittest.TestCase``". I would prefer to use test classes where they will clearly improve the code, and leave functions where they will not. I also think that less experienced python developers will have an easier time making tests if they only need to be functions that are named "test_*". 

In any case, I suggest that any test suite overhaul be done after the ``0.1`` release.

bmu, 2015-03-09:
My intention to move to Pandas TestCases is not to have "nicer code" or that we use classes without a need to.
The reason is, that these test cases really test a series or a DataFrame: they test for expected values and for the expected index.
This is especially useful because often time series are used as input for pvlib functions. My experience from everyday work with functions (not classes;-) that use Pandas and numpy is, that the preservation of Datetime indexes is sometimes a difficulty and can lead to unexpected results, that are not easy to detect.

Also for unexperienced programmers it may be usefull to have something like a template for testing, which could be provided by an existing test. 

No Problem with skipping this for now, but I think these Tests are really useful for future releases.

wholmgren, 2015-03-09:
That all makes sense. Good point regarding the templates too.

wholmgren, 2016-02-02:
A couple of things related to the stalled conversation above...

The nose documentation is now warning people that it's no longer in active development and recommends that new projects use pytest or just unittest. I've not used pytest, but it looks nice.

I added a bunch of ``assert_series_equal`` and ``assert_frame_equal`` statements to the test code, so I think that this covers most of @bmu's concerns in his last [comment](https://github.com/pvlib/pvlib-python/issues/12#issuecomment-77804841) and I wonder if this issue can be closed.

bmu, 2016-02-02:
So feel free to close it!

wholmgren, 2016-02-02:
thanks @bmu, closing.

closed 2016-02-02 by wholmgren