pvlib/pvlib-python #13: clearsky.py requires scipy.io

wholmgren, 2015-03-02:
Scipy either needs to be added to the setup.py file or we need to wrap the scipy statement with try/except and then throw a warning about reduced usability.

I would lean towards the latter as scipy installation is much more difficult than numpy and pandas. Best would be to remove the scipy.io dependency entirely.

bmu, 2015-03-02:
Recent matlab files should be hdf5 based (see [this question](http://stackoverflow.com/questions/874461/read-mat-files-in-python) on SO). So maybe we could simply update this file and read it with `pd.read_hdf` (Seems not to work with the current file, so maybe this is created with an older matlab version).

wholmgren, 2015-03-02:
Looks like it's matlab 5.

```$ head pvlib/data/LinkeTurbidities.mat 
MATLAB 5.0 MAT-file, Platform: PCWIN64, Created on: Mon Jul 16 10:16:10 2012
```

I checked the PVLIB 1.2 release and it's the same. I think that the ideal situation would be for both the matlab and python projects to use the same ``.h5`` file. Maybe this could be a part of the work going into #3.

Calama-Consulting, 2015-03-03:
Yes, I think the best would be for us to lead in translating this into a more portable file format, best case would be a .csv that can then be read with standard techniques (as this is just a 2D array). Any takers? In the near term, @wholmgren , could you add scipy.io to the setup.py to make sure we're internally consistent?

wholmgren, 2015-03-03:
Yes, but I would prefer to go with something like

```python
try: 
    import scipy.io
except ImportError:
    raise(ImportError, 'You must install scipy to automatically load the Linke Turbidities')
``` 

right before [line 137](https://github.com/pvlib/pvlib-python/blob/master/pvlib/clearsky.py#L137).

Calama-Consulting, 2015-03-12:
Sounds good to me

bmu, 2015-03-20:
I close this for 0.1 in favor of #36

closed 2015-03-20 by bmu