pvlib/pvlib-python #15: Initial release to PyPi

Calama-Consulting, 2015-03-03:
At what point should we push the initial release to PyPi? I'm leaning towards pushing after #6 and #13 are resolved with the 0.1 tag. 

Thoughts?

wholmgren, 2015-03-03:
Sounds good for the code. 

What should we do about the tutorials? My suggestion is to just promise that one good tutorial works at this point. I can fix up, or even remove, the module level ones later. Another idea is to make a new repo under the pvlib org with the tutorials (at least the detailed ones). Not sure if that would be confusing to people that are new to git/github/python. 

We should also double check that there are no big issues that we want to fix in the docs.

Calama-Consulting, 2015-03-03:
The tutorials on the dev branch of the sandia site are pretty good, and fixed for the new package layout, I'll port those over

wholmgren, 2015-03-08:
@Calama-Consulting do you have an approximate date in mind for the ``0.1`` release? I'll knock out a few of these easy issues this afternoon.

Calama-Consulting, 2015-03-08:
Let's aim for the end of this week, to give me time to make sure the tutorial are solid. We will also need to resolve #9, @wholmgren can you handle this as well? I beleive we had decided on the option with a modified `setup.py` which allows a user to manually install spa if they download the package from github and add their spa C code.

wholmgren, 2015-03-08:
Sounds good!

wholmgren, 2015-03-10:
Add #24 to 0.1 milestones.

wholmgren, 2015-03-11:
Looks pretty easy to [make a conda package](http://conda.pydata.org/docs/build_tutorials/pkgs.html). Should we do that for 0.1 or wait for a future release? I don't use Anacondas but it seems to be all the rage these days, especially for people that are new to Python e.g. former PVLIB_MATLAB users.

bmu, 2015-03-11:
+1 for a conda package. I use anaconda on Windows and linux. However I think its fine to wait for the next release. Maybe I could create the package than.

Calama-Consulting, 2015-03-12:
Sounds good, it doesn't look too involved,  I will add a conda wrapper and upload to binstar in addition to pypi for 0.1 . I think it will increase the ability to uptake, especially for new windows users.

dacoex, 2015-03-13:
Please ping me when you release the package to pypi.

I will adapt the Ubuntu package accordingly and propose a change in the docs for pointing at this repository:
https://code.launchpad.net/~pythonxy/+archive/ubuntu/pythonxy-devel/+packages?field.name_filter=pvlib

bmu, 2015-04-04:
What about the release?

wholmgren, 2015-04-14:
@Calama-Consulting what can we do to help get this out? I would like to get through 0.2 before writing the PVSC paper, and I think that requires pushing 0.1 as soon as possible.

Calama-Consulting, 2015-04-20:
Hi Folks, 

 Appologies for the delay on my end, I'm ready to push this out today, but want to confirm exactly which commit we want to mark as tag 0.1, and I'll take care of putting this up on pypi and conda.

wholmgren, 2015-04-20:
Great! I am good with tagging the current master as 0.1 or, if you have time for a quick PR review, after merging #51.

uvchik, 2015-04-20:
We mixed new features like spa_python (pvlib/pvlib-python@cc6a962) with bug fixes like pvlib/pvlib-python@70fb5c4. In future versions we should create a release branch a few weeks before the release date to separate bug fixes from new features.

For now I agree with @wholmgren to take the last commit (pvlib/pvlib-python@a50c8c6) or wait to integrate the pull requests (#50, #51).

Calama-Consulting, 2015-04-20:
Done with #51 , go ahead and pip install. Let me know if there are any issues, working on conda now

wholmgren, 2015-04-20:
I pip installed pvlib into clean Python 2.7 and 3.4 environments and ran the test suite. Everything is fine except for the failure in ``pvlib.test.test_pvsystem.test_retrieve_sam_network`` that is on all of our builds. I can download that file manually just fine, so I'm not sure what the issue is.

wholmgren, 2015-04-20:
Made the [0.1 docs](http://pvlib-python.readthedocs.org/en/0.1/). I forgot that we needed to update the 0.1 release date in the docs what's new section. So, we should change that in 0.2 (or 0.1.1?). 

Just realized that I am the only admin for the readthedocs page. I tried to add @Calama-Consulting and @bmu but you don't seem to have rtd accounts. So, I recommend that you go ahead and make the accounts, let me know if the username is different from your GH account, and I will add you to the rtd project.

bmu, 2015-04-22:
I created an account on rtd, username bmu.

bmu, 2015-04-22:
Can we close this issue?

wholmgren, 2015-04-22:
Yes, should we start a new one for 0.2? Wondering about the timeframe and scope for that.

wholmgren, 2015-04-22:
@bmu added you to the rtd project.

closed 2015-04-22 by wholmgren

uvchik, 2015-04-22:
Congratulation to all of us on our first release, but especially to the initiators and main contributors!

bmu, 2015-04-23:
I added a new issue for 0.2.