pvlib/pvlib-python #16: short term doc fixes

wholmgren, 2015-03-03:
A list of [doc](http://pvlib-python.readthedocs.org/en/latest/) fixes that we may consider before an initial release:

* Add more to the homepage, probably stuff from ``README.py``.
* Describe the relation to PVPMC, PVLIB-Matlab.
* Make a text link for the tutorials instead of my usual url dump.
* Different page for each module?

Any other relatively easy and important doc fixes?

jforbess, 2015-03-09:
Issues found in docs, likely from auto-gen content that hasn't been updated in code? Perhaps not all of these qualify as easy to fix before release: 

Ineichen doesn’t provide info on single df output
function is now aoi, but docs reference getaoi throughout, except in actual function doc
lots of references to pvl_function not function

wholmgren, 2015-03-09:
Thanks. 

The numpydoc convention is to avoid using "See also" unless it's really important. I think that we can follow that rule now now that the library structure is in better shape.

More issues:
* The doc title capitalization needs to be updated. 
* Copyright needs to be updated.
* ``pvlib.pvsystem.snlinverter`` has a poorly formatted table.
* References to ``maketimestruct, makelocationstruct`` in ``pvlib.clearsky.haurwitz``, maybe more.
* ``pvlib.irradiance.beam_component`` has no documentation.
* All modules need a better introduction to what's in them.
* ``pvlib.pvsystem.I_from_V`` has no documentation.
* ``pvlib.pvsystem.Voc_optfcn`` has no documentation.
* ``pvlib.pvsystem.calcparams_desoto`` says that ``module_parameters`` can be a ``Series``, which, although true, is probably confusing (does anybody really use a Series as a dict?). In any case, the doc's ``dict`` keys do not match the code's ``dict`` keys.
* Misspelling in ``pvlib.pvsystem.retrieve_sam`` example.
* Format equation in ``pvlib.pvsystem.singlediode``
* ``pvlib.pvsystem.systemdef`` as described in #17 
* ``pvlib.tmy.readtmy2`` refers to a metadata struct instead of a dict. Could be addressed in #7.

Calama-Consulting, 2015-03-13:
Closed from #22

closed 2015-03-13 by Calama-Consulting