pvlib/pvlib-python #28: docs: add tutorials to Sphinx build

dacoex, 2015-03-12:
Add
https://github.com/pvlib/pvlib-python/tree/master/docs/tutorials 
to
http://wholmgren-pvlib-python-new.readthedocs.org/en/docfixes/index.html#

maybe with a link to the nbviewer.
there's also a ipython sphinx plugin
http://matplotlib.org/sampledoc/extensions.html#ipython-sessions
https://github.com/ipython/ipython/tree/master/IPython/sphinxext

wholmgren, 2015-03-12:
There's already a link to an nbviewer page with the tutorials. 

"This documenation focuses on providing a reference for all of the modules and functions available in pvlib-python. For examples of how to use pvlib-python, please see the [tutorials](http://nbviewer.ipython.org/github/pvlib/pvlib-python/tree/master/docs/tutorials/)."

Inline code in documentation and IPython sphinxext is a separate issue.

closed 2015-03-12 by wholmgren

dacoex, 2015-03-12:
>There's already a link to an nbviewer page with the tutorials. 
Yes, but from which page in the docs at RTD can this be found?

wholmgren, 2015-03-12:
Not sure what exactly you want me to address here...

#22 is a PR to fix documentation problems raised in issue #16 (and more). This is what I quoted from, since this is what you linked to originally. 

The official documentation for pvlib-python is http://pvlib-python.readthedocs.org/en/latest/ and it too contains a link to the tutorials.

dacoex, 2015-03-12:
>Not sure what exactly you want me to address here...
Nevermin. overlooked that sentence 
"This documenation focuses on providing a reference for all of the modules and functions available in pvlib-python. For examples of how to use pvlib-python, please see the tutorials."

I was used to check only the ToC