pvlib/pvlib-python #29: add a io module

dacoex, 2015-03-12:
From:
https://github.com/pvlib/pvlib-python/pull/7#issuecomment-78441084

Shall we add standard readers for typical data providers such as:

Input:
*  Meteonorm
*  SolarGIS, e.g. http://solargis.info/doc/_files/sample_data/SolarGIS-climData/SolarGIS-TMY-hourly-P50-PSA-ES.zip
* BSRN
* SoDa-IS

Output:
* epw: http://apps1.eere.energy.gov/buildings/energyplus/cfm/weather_data3.cfm
* SAM

Hints:
https://github.com/pydata/pandas/blob/master/pandas/io/date_converters.py

This could be a meta issue for all input and output format existing in the (solar) energy meteo domain.

wholmgren, 2015-03-12:
I would support renaming ``tmy.py`` to ``io.py`` and adding importers for the most useful data sets. We need to be careful about the scope, though. A comment on the pandas issue correctly points out the danger

> It seems to me like this wiill either stagnate or grow into a melange of tailored
solutions to the 1001 weird data problems found in the wild that most pepole won't see.

It's also worth pointing out that pydata recently created a new repo [pandas-datareader](https://github.com/pydata/pandas-datareader) after having discussed this and related issues for a long time.

bmu, 2015-03-13:
I think we should call it `data_io`, because `io` is a module in the python standard library. Pandas uses `io` but if I remember correctly I already had some issues because of this name clash.

wholmgren, 2015-04-22:
Do we want to go forward with this for 0.2? I'm still concerned about scope and future maintenance. I don't regularly use these kinds of datasets, so I'm not a good judge of the tradeoffs. Wondering what @Calama-Consulting thinks of this.

Does ``pvsystem.retrieve_sam`` belong here too?

Not sure about naming. Technically the module would be named ``pvlib.<<something>>``, so shouldn't clashes should be fairly rare?

jforbess, 2015-04-22:
Agreed that future maintenance of dataset importing is an unknowable
effort.

I think the scope to consider here would be how to frame it so that someone
could easily make a reader for a common dataset and add it to pvlib,
because I don't think it makes much sense to make the readers in advance of
need, but it would be great to get people to contribute a good reader if
they found themself using SolarGIS a lot. Personally, I anticipate using a
lot of data from individual plants, without a standard form, but a
well-defined reader template might help with that as well. Not sure.

Not sure about the EPW output. Isn't that an input? It's an input to
PVsyst.

bmu, 2015-04-23:
One way for possible new readers would be to define a specific pvlib format for input and than every reader can read data and converts it to this format.

wholmgren, 2015-06-21:
I'm still interested in this but will leave it the community to resolve. It gets a "Someday" milestone until somebody expresses interest in resolving it.

dacoex, 2015-12-02:
@bmu Do you still think that this is needed?
Where do we get sample data for above data sources?
Which location is good?
Carpentras?
Or any where we get data for?

bmu, 2015-12-03:
There is no need from my side at least ;-). So maybe Wills solution to let users add some readers is the best one.

Dont't think that it is a problem to geht some sample data. I could add Meteonorm data (and ask for permission)  as well as SolarGis data.

dacoex, 2015-12-03:
>There is no need from my side at least ;-). So maybe Wills solution to let users add some readers >is the best one.

OK, so we close the issue and then let users individually create the dateconverter functions etc.?

wholmgren, 2015-12-03:
@dacoex I don't have anything new to add to the discussion and it's fine with me if we close it. You created the issue so you can close it if you think we've reach a resolution.

dacoex, 2016-01-05:
Here's a repo for some sample data (still to be filled):
https://github.com/dacoex/pvlib_data

I plan to add some data for testing of the io functions (to be coded).

Additionally, the data can be used for tutorials / notebooks / cross-comparison with other tools / etc.

statsmodels do this for 

* Stata: https://github.com/statsmodels/statsmodels/blob/master/statsmodels/stats/tests/test_panel_robustcov.py
* R Project https://github.com/statsmodels/statsmodels/blob/master/statsmodels/stats/tests/gen_contingency_table_tests.R

cwhanse, 2016-11-08:
Will, others,

I'd like to re-open this issue and volunteer for the task of creating the .io module.  I think we should focus on data import, providing tools to bring data into pvlib applications.

I'm going to start with importing data from the Surfrad archive, not that its the most needed converter, but rather because I already have some code to read these files.  The first major task for pvlib is delivering the content in pvlib conventions.

Cliff

adriesse, 2016-11-09:
Your update led me to the read the foregoing comments.  I have spent a lot of time over the years importing, exporting, converting and even reverse-engineering data files.  Any import routine that is published in pvlib is likely to find a thankful user somewhere down the line.  Maybe even me.

I just wonder, would pecos be a better place for such tools?

dacoex, 2016-11-09:
@cwhanse I could started this issue and prepared even a sample test data repo ([pvlib_data](https://github.com/dacoex/pvlib_data)) which be moved under the project umbrella (https://github.com/pvlib)

I have a reader function for PVSyst output & MACC-RAD. 
We would have agree on the way we to reformat the output. Or are you simply looking for a short cut for importing the raw data into a dataframe?

@adriesse I would prefer to have meteo / solar / radiation specific formats in a module. As I understand pecos it's more of a processing tool using other support libs.

dacoex, 2016-11-14:
See also further discussion on design in:
https://github.com/pvlib/pvlib-python/issues/261

PR #272 (2018-09-25): make iotools package
I made a new ``io`` branch to try to streamline the process of multiple users contributing their new io tools and to make it easier for all of us to test it them without hurting the master branch. I also went ahead and moved the tmy readers into the package and updated the tests accordingly. I named the package ``iotools`` since Cliff expressed a preference for that name, but it's not yet set in stone. I think this is mostly consistent with what we talked about in #261 and #29. Let me know if I missed something concerning the structure.

Please make your new IO tools pull  [... cut]
closed 2018-09-25 by wholmgren