pvlib/pvlib-python #30: Port Tutorials

Calama-Consulting, 2015-03-12:
Port of tutorials from sandia labs develop branch.

bmu, 2015-04-04:
I think we could shift this to 0.2 and release 0.1 of pvlib.

wholmgren, 2015-04-05:
Agreed.

bmu, 2015-04-06:
@Calama-Consulting ?

bmu, 2015-04-22:
I changed the milestone to 0.2

wholmgren, 2015-06-25:
I'm going to mark this 0.3 since I'm not sure what should be ported into new or existing notebooks and what should be put into the docstrings and sphinx documentation.

wholmgren, 2015-07-09:
I've had a lot of offline feedback that broken tutorials a big problem and I'd like to do a 0.2.1 release with updated tutorials. I don't plan to make them read any better, just get them to the point that they can run end-to-end without exceptions. #69 is also relevant to this.

NelisW, 2015-07-11:
Will try to help here. My take on the Sphinx vs notebook issue is that personally I prefer a worked example to get up to speed first and then dig into the API-level docs. Trying to get into a new product just by reading the API docs is just too hard. So, let's fix the notebooks first.

PR #74 (2015-07-12): Tutorials
~~Not yet ready to merge~~ 

Closes #30. This PR is just about making the notebooks work from beginning to end. I'm not going to spend much time making them read better.

I wrote a new [``tmy_to_power``](http://nbviewer.ipython.org/github/wholmgren/pvlib-python/blob/tutorials/docs/tutorials/tmy_to_power.ipynb) notebook that replaces ``Test_Scripts``. 

Here's a nbviewer [link to my fork](http://nbviewer.ipython.org/github/wholmgren/pvlib-python/tree/tutorials/docs/tutorials/).

To do:
- [x] ET_Irradiance (move into irradiance)
- [x] TS_2_Irradiance_functions reworked and renamed tmy_ [... cut]
closed 2015-07-12 by bmu