pvlib/pvlib-python #40: solarposition.ephemeris

caro-rli, 2015-03-24:
returns solar azimuth angle in wrong definition with zero = south. But irradiance.aoi needs as input panel azimuth angles from north.

This function is discouraged anyway, so may it be better to remove it from the library?

wholmgren, 2015-03-24:
I'm inclined to say that we should fix it, but I'm ok with removing it too. If I remember right, @Calama-Consulting is in favor of keeping it. A PR (with tests) is welcome.

wholmgren, 2015-03-26:
We'll keep using this issue to solve the timezone problem in this function.

The pyephem and NREL spa functions use UTC, so as long as Python, pandas, pytz, and dateutils can get the conversion right, DST should be ok there. ``test_solarposition.py`` has several tests that use a Golden, CO ``Location`` with a DST-allowing ``America/Denver`` timezone.

See #41 for a summary of the tz issue.

wholmgren, 2015-03-26:
This code block causes problems.

```python
    # this code is needed to handle the new location.tz format string
    try:
        utc_offset = pytz.timezone(location.tz).utcoffset(
            time[0]).total_seconds() / 3600.
    except ValueError:
        utc_offset = time.tzinfo.utcoffset(time[0]).total_seconds() / 3600.
    pvl_logger.debug('utc_offset={}'.format(utc_offset))
```

It creates a fixed ``utc_offset`` based the first element of the DatetimeIndex. ``utc_offset`` should instead be an array so that it can handle a change from DST to non-DST. I believe that the original code suffers from the same problem.

To fix it, I suggest using [``pvlib.tools.localize``](https://github.com/pvlib/pvlib-python/blob/master/pvlib/tools.py#L106), and the ``hours`` attribute of the returned object in place of ``UnivHr = DecHours + utc_offset - .5``.

I think this also removes the need for

```python
    Hour = time.hour
    Minute = time.minute
    Second = time.second
```

wholmgren, 2015-03-26:
cc @uvchik.

uvchik, 2015-03-31:
The Problem is, that I don't really understand the function and it is not well commented. I tried some thing but I didn't succeeded. 

I think I won't be able to repair and test this function in an acceptable period. First I have to learn more about Pandas timestamps and how to work with pandas and time zones.

Shall we add another warning, that using this function is on one owns risk. Or does anyone have an idea how to fix an test it?

wholmgren, 2015-03-31:
Thanks for looking. I'll see if I can implement my thoughts above.

PR #41 (2015-03-26): Fixed inconsistent azimuth angle in the ephemeris function
Fixed #40.
PR #45 (2015-04-04): fix built in ephemeris
I fixed the ephemeris function (I think). The key changes are

* replace the local and utc hour calculation code as described in #40.
* multiply longitude by -1 as described in a comment but not actually done in the code.

Here's some sample output.

Tucson summer elevation
[image]

Berlin no DST (March 28, 2015)
[image]

Berlin DST (March 30, 2015)
[image]

Berlin summer elevation and azimuth.
[image]
[image]

Closes #40. cc @uvchik
closed 2015-04-04 by bmu