pvlib/pvlib-python #46: Contributing

bmu, 2015-04-04:
If we agree on it, we could either completely rely on [pandas](http://pandas-docs.github.io/pandas-docs-travis/contributing.html) or give some rules, where we use our own conventions (in the [wiki](https://github.com/pvlib/pvlib-python/wiki/Development-information) or in the docs).

wholmgren, 2015-04-05:
Most of the stuff I wrote was inspired by pandas (maybe I should have made that more clear), but I felt that we should have something more than just "go look at pandas" since new contributors might not yet understand the scientific python ecosystem. I'm open to changing this, though.

bmu, 2015-04-06:
Ok, I think this is a good approach. However we could provide the link to pandas, if we agree on all of the thinks that are stated there.

wholmgren, 2015-04-06:
I added a handful of links to the pandas pages. Feel free to edit anything.

bmu, 2015-04-06:
Ok, thanks. I'll close this issue.

wholmgren, 2015-05-11:
The new 0.16.1 pandas release has a new [Contributing to pandas](http://pandas.pydata.org/pandas-docs/stable/contributing.html) page. It's really good. I still think we should have a small wiki, but I'll add and emphasize links to this.

wholmgren, 2015-06-21:
I renamed the *Development information* wiki to *Contributing to pvlib python* and rewrote some of the text. I hope we can get some other people to add-to/clarify the wiki in the future, so I'm going to leave this open for 0.3. I think @jforbess's perspective would be helpful here. I found it hard to strike a balance between high-level overview with links as opposed to detailed instructions.

PR #135 (2016-03-16): documentation updates
A handful of documentation updates...

- [x] closes #46: contributing instructions
- [x] closes #73: how to cite pvlib
- [x] add pvlib installation instructions to docs
- [x] closes #116: update comparison to pvlib matlab
- [x] closes #47: add a brief page on how pvlib treats time and timezones
- [x] update whatsnew

http://wholmgren-pvlib-python-new.readthedocs.org/en/contributing/
closed 2016-03-16 by wholmgren