pvlib/pvlib-python #49: The documentation is broken

uvchik, 2015-04-07:
The documentation is broken:
http://pvlib-python.readthedocs.org/en/latest/pvlib.html

I tried to compile it myself but I get import errors from sphinx, which I don't understand. Maybe for the same reason as the build on readthedocs.

Example:
The import of tools (`from pvlib import tools`) works fine for me but sphinx raises the following:
```python
.../pvlib-python/docs/sphinx/source/pvlib.rst:23: WARNING: autodoc: failed to import module u'pvlib.irradiance'; the following exception was raised:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/sphinx/ext/autodoc.py", line 335, in import_object
    __import__(self.modname)
  File "/home/uwe/rli-lokal/git_home/pvlib-python/pvlib/__init__.py", line 4, in <module>
    from pvlib import tools
ImportError: cannot import name tools
```

wholmgren, 2015-04-07:
Thanks, I'll look into it.

wholmgren, 2015-04-07:
The problem was that I added a ``from ... import ..`` that wasn't captured by ``conf.py``'s Mock statements. Seemed easier to just remove the ``from``. We'll try to be better about checking documentation builds in the future.

Closed by 70fb5c4de.

bmu, 2015-04-23:
It doesn't work for me locally. I created a pull request to fix it, see #56.

I think the problem is, that autodoc can't find the modules (see the `ImportError` above, same here).
It may find it, if pvlib is installed, however one needs the modules from the current repository if you want to build the current documentation.

PR #56 (2015-04-23): Fix sphinx
should fix #49
closed 2015-04-23 by wholmgren