pvlib/pvlib-python #52: Current build failing on `test_retrieve_sam_network`

Calama-Consulting, 2015-04-20:
It looks like this is related to the downloads from the NREL website. When I tried to manually download the files, it took about a minute for the first download to happen (and it was fine after that). We might want to consider hosting the files locally? I'm not sure what the limitations are on that.

wholmgren, 2015-04-20:
The files total only 2 MB, so to me it doesn't seem like a big deal to include them along with the Linke turbidities and eventually the Driesse inverter database. This is what the Matlab version does too.

Calama-Consulting, 2015-04-21:
I ran it again locally on my machine, and it worked fine. I also re-started the travisCI and we're now passing. So, I think this is fine for now. The advantage of pulling from SAM is that the database is always up to date, however a future addition might be to have a set of files included as a baseline to ensure that a user can use these algorithms without an internet connection, or if the website is down.

wholmgren, 2015-04-21:
The links are actually to specific versions of the files (see [here](https://github.com/pvlib/pvlib-python/blob/fb7c1c6a2388b04102197825650a578b614a50c4/pvlib/pvsystem.py#L533)).

bmu, 2015-04-23:
Is this resolved? I am trying to close the 0.1 milestone ;-)

wholmgren, 2015-04-23:
Well, it's resolved in the sense that the test passes again, but there is still the question of if we should just include the files in the distribution. Another possibility is to remove this test from the standard test suite. Those all seem like 0.2 questions, though.

wholmgren, 2015-04-27:
The same test failed again, though in a slightly different way and for only one of the builds. 

https://travis-ci.org/pvlib/pvlib-python/jobs/60250664

Calama-Consulting, 2015-04-27:
I think it would be worthwhile to include the files locally, I'm putting together a PR

wholmgren, 2015-06-21:
I'm going to push this to 0.3.

PR #224 (2016-07-22): add sam files to distribution
This PR adds the SAM files to the distribution and makes minor changes to the pvsystem.retrieve_sam API. See the Files Changed for more information.

I'll merge tomorrow unless there are comments.

Closes #52.
closed 2016-07-22 by wholmgren