pvlib/pvlib-python #55: Release 0.2

bmu, 2015-04-23:
This issue is for discussion of the 0.2 release process.

bmu, 2015-04-23:
For now, the date of the 0.2 milestone is set to April 30 2015. However I think this is not realistic.
Any suggestions?

wholmgren, 2015-04-23:
Depends on if we want to keep everything currently labelled as 0.2. I think that 4 weeks a good amount of time to accomplish #2, #20, #30, #37, #42, #54, and #47. In addition, the testing issues should be at least partially addressed.

#29, #39, and #17 seem like they're going to require more thought and discussion to do well, so I think we'll need to limit the initial scope of those issues, label those parts of the library as experimental, or push them off.

wholmgren, 2015-06-19:
We got a lot of positive feedback on the project at PVSC. The most common issue was installation, especially for new users on Windows. I want to capitalize on whatever extra interest we get out of PVSC, so, I'd like to resolve the easier issues and then release 0.2 with a conda package on Monday or Tuesday. I am going to devote some time today and over the weekend, and then we'll go with whatever state the project is in. I think I can knock out the list above except for #2.

Any objections @bmu @Calama-Consulting or others?

If nobody speaks up then I will break the rules and accept my own PRs.

wholmgren, 2015-06-20:
I went ahead and made conda packages for 0.1 and updated the readme. Here's the process that I stumbled into https://github.com/pvlib/pvlib-python/wiki/Release-procedures

bmu, 2015-06-21:
I can have a look at your PR's and merge them. However, I have to focus on my PhD in my spare time and it seems, that I won't get any working time to contribute to pvlib. So my own contribution will not be to much. Maybe we can shift some 0.2 issues to "someday" and release it as it is. I will start for some of "my" issues.

wholmgren, 2015-06-21:
Great thanks @bmu.

wholmgren, 2015-07-06:
Ok, my suggestion is that we go forward with the 0.2 release. We can make a 0.2.1 with tutorial improvements if @NelisW, @Calama-Consulting, or myself find the time.

wholmgren, 2015-07-06:
@Calama-Consulting can you upload to PyPI if you think we're ready? I updated the whats new date and the version.

wholmgren, 2015-07-08:
Closed by 263c90ef26

closed 2015-07-08 by wholmgren