pvlib/pvlib-python #60: doc: describe modeling steps

dacoex, 2015-04-25:
Just to catch your options:

Would the documentation benefit from having a section, mapping the functionality against the typical modelling steps outlined in:
https://pvpmc.sandia.gov/modeling-steps/
https://pvpmc.sandia.gov/wp-content/uploads/2014/08/1199X499_slide2.png

If so, I would embark (slowly) on that.
It would help me and maybe others to understand the library.

wholmgren, 2015-04-27:
Better linking to PVPMC would be very helpful, but we should avoid duplicating their effort. Better ipython notebook tutorials should be a priority too (see #30). I think we can make the equivalent of ``pvlib_functions_by_cat.html`` by modifying the sphinx scripts.

Would be happy to see some PRs for documentation.

dacoex, 2015-04-28:
>I think we can make the equivalent of pvlib_functions_by_cat.html by modifying the sphinx scripts.

OK. that's it. I will have a look at this someday ;-)

dacoex, 2016-01-25:
I think the following will solve this:

* https://github.com/wholmgren/pvlib-python/blob/pvsystem/docs/sphinx/source/package_overview.rst
* any further modelchain documentation