pvlib/pvlib-python #61: meta: add Shading, Soiling, and Reflection Losses

dacoex, 2015-04-25:
These seem to miss on the environmental input side:
https://pvpmc.sandia.gov/modeling-steps/1-weather-design-inputs/shading-soiling-and-reflection-losses/

Maybe this meta issue can track this as mentioned in 
https://github.com/pvlib/pvlib-python/issues/2#issuecomment-74643819

wholmgren, 2015-04-27:
PRs for shading and soiling models are welcome. To my knowledge, this is not something that's currently in the Matlab project.

There are already two reflection loss functions: ``pvsystem.physicaliam`` and ``pvsystem.ashraeiam``. Did you want something different?

cwhanse, 2020-08-20:
I think this can be closed as a general issue, with #860 #669 #1017 and all of `pvlib.iam`.

closed 2020-08-20 by wholmgren