pvlib/pvlib-python #69: IPython notebooks update

NelisW, 2015-06-21:
I have started work on bringing the notebooks in line with the latest code.

The first thing I ran into was the issue of versions.  Is the following ok with your versions?

Python	2.7.8 32bit [MSC v.1500 32 bit (Intel)]
IPython	3.0.0
numpy	1.9.2
scipy	0.15.1
matplotlib	1.4.3
pandas	0.15.2
pvlib	0.2.0dev

wholmgren, 2015-06-21:
Great, we really need help improving the tutorials. I think that we especially need good some good library-level tutorials (as opposed to the existing module-level tutorials).

Not sure the exact issue is. Are you asking what versions the tutorials should already work with or what versions the tutorials should be made compatible with or something else? In any case, there's not currently a standard, but I will propose one here:

* The latest version of pvlib
* The requirements in ``setup.py``
* Seaborn can be used if protected with try/except in the import. There are a few cases where I've used a Seaborn call later in the tutorials. I'm ok with this as long as there's a good reason for it.
* IPython 3 is ok with me.

One warning: I'm about to start working on applying the [Variables and style rules](https://github.com/pvlib/pvlib-python/wiki/Variables-and-style-rules) as discussed in #37, and this is going to break a lot of code and tutorials. Can you say a little more about what specifically you were going to work on and what your timeline is? I'm grateful for the help, I just want to avoid duplicating effort.

NelisW, 2015-06-22:
Hi Will - thanks for your efforts and keep up the good work!

I just only started yesterday with PV. I want to install a PV system in my home and want to make sure that it makes financial sense.  We are only going to stay on five to eight years in this home before moving to the coast - so I want to understand in some detail what benefit the system will have to me over the next few years.

I come from an optical radiometry background; wrote a book on radiometric modelling and computational radiometry with SPIE and developed a Python toolkit for radiometry calculations and some IPython notebooks as tutorials.
http://spie.org/Publications/Book/2021423?origin_id=x646
https://github.com/NelisW/pyradi
http://nelisw.github.io/pyradi-docs/_build/html/index.html
https://github.com/NelisW/ComputationalRadiometry#computational-optical-radiometry-with-pyradi
Also wrote a IPython notebook to LaTeX compiler:
https://github.com/NelisW/ipynb2tex

Areas that I can contribute in is in (1) atmospheric models (I have access to and use Modtran extensively) (2) pvlib Python coding, documentation, and (3) notebooks.  I am quite willing to publish my end-to-end modelling work for a small home system (if I can get that far!).

I wanted to get to grips with pvlib, so starting with the IPython tutorials makes sense.  My questions really stem from problems with different versions of tools used by different people in the team. If I push IPython 3 notebooks  (nbformat=4 http://ipython.org/ipython-doc/3/notebook/nbformat.html#nbformat) and you are using IPython 2/nbformat3 tools you will not be able to read the new notebooks.

    [*The latest version of pvlib*]  Yes, I am working from github
    [*The requirements in setup.py*] thanks will look into this
    [*Seaborn can be used*] OK, will follow this. 
    [*IPython 3 is ok with me.*] thanks, great relief that I don't have to convert back to 2

I am considering a small inverter from IMEON http://www.imeon-energy.com/en/imeon-3-6/ - still lokking for their inverter model parameters. I asked them to please supply.

I am living in Pretoria,  South Africa - trying to find TMY3 data for my area.  The closest I could find was Harare, which has quite a different cloud behaviour.

jforbess, 2015-06-22:
Nelis, it's great to have you join pvlib.

http://irena.masdar.ac.ae might have the solar data you need, though I find
the site pretty unintuitive, and focusing on mapping annual averages rather
than providing hourly data. I haven't dug very far in.

It may be bad form to suggest other solar tools right here, but I would
recommend SAM (http://sam.nrel.gov) as a good point of reference for a
modeling tool, to help you identify potential system losses that may not be
fully considered in pvlib at the moment.

Jessica

wholmgren, 2015-06-22:
Thanks Nelis. I never considered an individual homeowner as a use-case for pvlib, but I'm glad to hear of it. It would be great to see an IPython notebook for that. I agree with @jforbess's suggestion to check out SAM. 

Your radiometry tools look nice and I'm going to dig into them further once we get 0.2 released. Good spectral correction tools would be a very nice addition to pvlib and could really help certain parts of the PV community. I don't have any experience with this topic though, and I hope that somebody else can lead it.

bmu, 2015-06-22:
Another option is [PVGIS](http://re.jrc.ec.europa.eu/pvgis/apps4/pvest.php?map=africa), which is very simple, but gives a good indication, what you can expect. But I also think, a pvlib notebook for such a case would be nice.

Calama-Consulting, 2015-06-22:
Nelis, nice to have you on board! For the conversation on spectral integration, I just added some info on a package I worked on a while ago in #71 , I'd be interested in hearing your comments for improvements and integration in that thread! Your pyradi package is outstanding, and your tutorial notebooks are now the tools I will be using to draw new folks into python.

dacoex, 2015-06-22:
@NelisW 

The closes to SA I know off my head:
http://www.crses.sun.ac.za/files/research/publications/technical-reports/GeoModelSolar_SolarResRep_58-01-2011_Upington_rev2.pdf

Not quite Gauteng but...

NelisW, 2015-06-23:
Hi All!
Really nice to receive such warm welcome - must be the sun in your PVs!  Thanks!!

thanks for your hints and suggestions for models and data sources, I followed up and found a university network http://www.sauran.net/. The data does not go back a long time but is quite useful for my purpose.

PyRTM & SMARTS: I haven't used SMARTS before, done most of my work in Modtran. But yes, I should be able to contribute in this area.  First I just need to dig in to pvlib :-)  This is a part-time hobby, so the going might be slower than I would like.

regards

nelis

NelisW, 2015-06-23:
Thanks to all with the feedback (within 48 hours!) we can now close this one.

closed 2015-06-23 by NelisW