pvlib/pvlib-python #83: pvsystem.py - no continuous IV curve data

cbirkj, 2015-07-27:
The singlediode function, located in pvsystem.py, does not have the capability to return continuous current and voltage values.  I have modified the code to include a loop:

si = np.ones((NumPoints,1))
sv = np.linspace(-1,1,NumPoints)
if NumPoints >= 2:
    	for i in range(NumPoints):
    		V[i,:] = Voc.values * sv[i]
    		I[i,:] = I_from_V(Rsh=Rsh, Rs=Rs, nNsVth=nNsVth, V=V[i,:], I0=I0, IL=IL)

This may not be the most efficient way but it worked.

- Birk

wholmgren, 2015-07-28:
Thanks, Birk. 

I'm not sure how to best construct the API for this. I removed the ``kwargs``, and thus ``NumPoints``, from the from the ``singlediode`` function in the 0.2 API refactor. One approach is to use two different functions, one for labelled points and one for arbitrary points. What are you returning from the function? A dataframe and, optionally, a series? You could just embed the whole function in a comment using [triple ticks and the python declaration](https://help.github.com/articles/github-flavored-markdown/#syntax-highlighting).

Also, I think that ``i_from_v`` will work with vector inputs, and this will be a lot faster than iterating over each point.

What do others think?

wholmgren, 2016-06-13:
Here's my code to make IV curves. I'm still not sure if any of this should be included in pvlib or not. I think we should at least add something similar to this to the documentation. Maybe someone else has a better way to do this.

```python

def fivepoints_to_frame(pnt):
    """
    Converts a 1 dimensional, dict-like singlediode or sapm result
    to a 5 row DataFrame with columns current and voltage.
    Users can iterate over the rows of a multidimensional
    singlediode or sapm result, if necessary.
    """
    ivframe = {'i_sc': (pnt['i_sc'], 0),
               'p_mp': (pnt['i_mp'], pnt['v_mp']),
               'i_x': (pnt['i_x'], 0.5*pnt['v_oc']),
               'i_xx': (pnt['i_xx'], 0.5*(pnt['v_oc']+pnt['v_mp'])),
               'v_oc': (0, pnt['v_oc'])}
    ivframe = pd.DataFrame(ivframe, index=['current', 'voltage']).T
    ivframe = ivframe.sort_values(by='voltage')
    
    return ivframe

resistance_shunt = 16
resistance_series = 0.094
nNsVth = 0.473
saturation_current = 1.943e-09
photocurrent = 7
module_parameters = pvsystem.retrieve_sam('cecmod')['Example_Module']

v_oc = pvsystem.v_from_i(resistance_shunt, resistance_series, nNsVth, 0, saturation_current, photocurrent)
voltage = np.linspace(0, v_oc, 100)

current = pvsystem.i_from_v(resistance_shunt, resistance_series, nNsVth, voltage, saturation_current, photocurrent)

fivepnts = pvsystem.singlediode(
    module_parameters, photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth)
ivframe = fivepoints_to_frame(fivepnts)

fig, ax = plt.subplots()
ax.plot(voltage, current)
ax.scatter(ivframe['voltage'], ivframe['current'], c='k', s=36, zorder=10)
ax.set_xlim(0, None)
ax.set_ylim(0, None)
ax.set_ylabel('current (A)')
ax.set_xlabel('voltage (V)')
```
[image]

The ``fivepoints_to_frame`` function takes this input

```python
fivepnts
{'i_mp': 6.104597093609919,
 'i_sc': 6.9584938427061607,
 'i_x': 6.6367680322891767,
 'i_xx': 4.4589886731292712,
 'p_mp': 51.257040864030472,
 'v_mp': 8.3964658237125214,
 'v_oc': 10.362416515927016}
```

and creates this output

```python
print(ivframe)
       current    voltage
i_sc  6.958494   0.000000
i_x   6.636768   5.181208
p_mp  6.104597   8.396466
i_xx  4.458989   9.379441
v_oc  0.000000  10.362417
```

requires #190.

cwhanse, 2016-06-13:
Will,

I’d include the code, but I’d also have the function return a second dataframe with all 100 computed points.

Cliff

[ivcurve]<https://cloud.githubusercontent.com/assets/4383303/16017889/d7dfd472-3156-11e6-8f61-315051f7ebdb.png>

wholmgren, 2016-06-13:
Another option is to reproduce the pvl_singlediode.m Matlab API by switching our singlediode's output from a DataFrame a simple dictionary. The dictionary values would be scalars or arrays as determined by the numpy broadcasting rules.

Here's another try at the multiple function approach. We'd have the existing ``singlediode`` that works with either scalar or time series inputs, and these functions that handle IV curve stuff for scalar inputs. 

I don't know if one approach is better than the other.

```python
def singlediode_points_to_frame(pnt):
    """
    Converts a dict of scalars describing the 5 points of
    a single diode curve to a 5 row DataFrame with
    columns current and voltage.

    Useful for plotting.
    
    Parameters
    ----------
    pnt: dict
        Typically the output of the singlediode function.
        Must have keys i_sc, p_mp, i_x, i_xx, v_oc.
        
    Returns
    -------
    fivepnts: DataFrame
        Columns labeled voltage, current.
        Rows labeled i_sc, p_mp, i_x, i_xx, v_oc.
    """
    fivepnts = {'i_sc': (pnt['i_sc'], 0),
                'p_mp': (pnt['i_mp'], pnt['v_mp']),
                'i_x': (pnt['i_x'], 0.5*pnt['v_oc']),
                'i_xx': (pnt['i_xx'], 0.5*(pnt['v_oc']+pnt['v_mp'])),
                'v_oc': (0, pnt['v_oc'])}
    fivepnts = pd.DataFrame(fivepnts, index=['current', 'voltage']).T
    fivepnts = fivepnts.sort_values(by='voltage')
    
    return fivepnts

def singlediode_ivcurve(module_parameters, photocurrent, saturation_current,
                        resistance_series, resistance_shunt, nNsVth, v_oc=None, num=50):
    """
    Creates a single diode IV curve with uniformly spaced points.
    
    Parameters
    ----------
    See singlediode
    
    v_oc: None or float
        Will be calculated if None.
    num: int
        Number of IV curve samples.
        
    Returns
    -------
    ivcurve: DataFrame
        Columns labeled voltage, current.
        Rows labeled by point number.
    """
    
    if v_oc is None:
        v_oc = pvsystem.v_from_i(
            resistance_shunt, resistance_series, nNsVth, 0, saturation_current, photocurrent)
        
    voltage = np.linspace(0, v_oc, num)
    current = pvsystem.i_from_v(
        resistance_shunt, resistance_series, nNsVth, voltage, saturation_current, photocurrent)
    ivcurve = pd.DataFrame(np.array([voltage, current]).T, columns=['voltage', 'current'])
    
    return ivcurve

def singlediode_points_ivcurve(module_parameters, photocurrent, saturation_current,
                               resistance_series, resistance_shunt, nNsVth,
                               num=50):
    """
    Solves the single diode model for scalar inputs and returns an IV curve
    and the 5 IV curve characterization points.
    
    Parameters
    ----------
    see singlediode
    
    Returns
    -------
    fivepnts: DataFrame
        Columns labeled voltage, current.
        Rows labeled i_sc, p_mp, i_x, i_xx, v_oc.

    ivcurve: DataFrame
        Columns labeled voltage, current.
        Rows labeled by point number.
    """
    
    fivepnts = pvsystem.singlediode(
        module_parameters, photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth)
    fivepnts = singlediode_points_to_frame(fivepnts)

    ivcurve = singlediode_ivcurve(
        module_parameters, photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth,
        v_oc=fivepnts.ix['v_oc', 'voltage'])
    
    return fivepnts, ivcurve

resistance_shunt = 16
resistance_series = 0.094
nNsVth = 0.473
saturation_current = 1.943e-09
photocurrent = 7

fivepnts, ivcurve = singlediode_points_ivcurve(
    module_parameters, photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth)

fig, ax = plt.subplots()
ax.plot(ivcurve['voltage'], ivcurve['current'])
ax.scatter(fivepnts['voltage'], fivepnts['current'], c='k', s=36, zorder=10)
ax.set_xlim(0, None)
ax.set_ylim(0, None)
ax.set_ylabel('current (A)')
ax.set_xlabel('voltage (V)')
# as shown above
```

cwhanse, 2016-06-14:
I would prefer the dict output rather than the dataframe for IV curves.  We won’t use the pandas time series functionality for IV curves so the extra overhead doesn’t make sense to me.

wholmgren, 2016-06-14:
People do use the singlediode function for time series analysis, though, so I don't want to take that away from them (or at least me). I think everyone can be happy, and very little code should break, if I amend my previous comment to say:

The dictionary values would be scalars, arrays, or *series* as determined by the input type.

See below for a couple of API proposals. Option 1 can be implemented immediately. Option 2 would break existing code. 

Option 3 (not shown), would use option 1 and create a separate "tuplizer" function. The tuples are nice for plotting.

```python
def singlediode(module, photocurrent, saturation_current,
                resistance_series, resistance_shunt, nNsVth,
                ivcurve_pnts=None):
    """
    Parameters
    ----------
    ...
    
    ivcurve_pnts: None or int
        The number of points with which to simulate an IV curve.
        If None or 0, an IV curve will not be generated.
    
    Returns option 1
    -------
    output: dict
        A dict with the following keys:
        
        * i_sc -  short circuit current in amperes.
        * v_oc -  open circuit voltage in volts.
        * i_mp -  current at maximum power point in amperes.
        * v_mp -  voltage at maximum power point in volts.
        * p_mp -  power at maximum power point in watts.
        * i_x -  current, in amperes, at ``v = 0.5*v_oc``.
        * i_xx -  current, in amperes, at ``V = 0.5*(v_oc+v_mp)``.
        * i - None or iv curve current. I'd also consider an empty array or missing key instead of None. 
        * v - None or iv curve voltage. I'd also consider an empty array or missing key instead of None. 
        
        
    Returns option 2
    -------
    output: dict
        A dict with the following keys and (voltage, current, power) values:
        
        * i_sc -  (0, i_sc, 0)
        * v_oc -  (v_oc, 0, 0)
        * p_mp -  (v_mp, i_mp, p_mp)
        * i_x -  (v_x, i_x, p_x) at ``v_x = 0.5*v_oc``
        * i_xx -  (v_xx, i_xx, p_xx) at ``v_xx = 0.5*(v_oc+v_mp)``
        * ivcurve - None or (voltage, current, power) tuple. I'd also consider a tuple of empty arrays or a missing key instead of None. 
    """
```

cwhanse, 2016-06-14:
What I meant, and perhaps didn’t say clearly, is that a dataframe (time series object) doesn’t seem a natural fit to contain a single IV curve.  A time series of a point on IV curves (e.g., Isc) makes a lot of sense.

Option 1 is better than 2, I think.

closed 2016-07-22 by wholmgren

evgenyna, 2017-06-04:
import pandas as pd
import numpy as np
import matplotlib.pyplot as plt
from pvlib.pvsystem import singlediode, v_from_i, i_from_v, retrieve_sam

def fivepoints_to_frame(pnt):
    """
    Converts a 1 dimensional, dict-like singlediode or sapm result
    to a 5 row DataFrame with columns current and voltage.
    Users can iterate over the rows of a multidimensional
    singlediode or sapm result, if necessary.
    """
    ivframe = {'i_sc': (pnt['i_sc'], 0),
               'p_mp': (pnt['i_mp'], pnt['v_mp']),
               'i_x': (pnt['i_x'], 0.5*pnt['v_oc']),
               'i_xx': (pnt['i_xx'], 0.5*(pnt['v_oc']+pnt['v_mp'])),
               'v_oc': (0, pnt['v_oc'])}
    ivframe = pd.DataFrame(ivframe, index=['current', 'voltage']).T
    ivframe = ivframe.sort_values(by='voltage')

    return ivframe

resistance_shunt = 16
resistance_series = 0.094
nNsVth = 0.473
saturation_current = 1.943e-09
photocurrent = 7
module_parameters = retrieve_sam('cecmod')['Example_Module']

v_oc = v_from_i(resistance_shunt, resistance_series, nNsVth, 0, saturation_current, photocurrent)
voltage = np.linspace(0, v_oc, 100)

current = i_from_v(resistance_shunt, resistance_series, nNsVth, voltage, saturation_current, photocurrent)

fivepnts = singlediode(
    module_parameters, photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth)
ivframe = fivepoints_to_frame(fivepnts)

fig, ax = plt.subplots()
ax.plot(voltage, current)
ax.scatter(ivframe['voltage'], ivframe['current'], c='k', s=36, zorder=10)
ax.set_xlim(0, None)
ax.set_ylim(0, None)
ax.set_ylabel('current (A)')
ax.set_xlabel('voltage (V)')

Hi, Im using your cide but getting error: 
"in i_from_v
    np.exp(Rsh*(Rs*(IL+I0)+V) / (nNsVth*(Rs+Rsh))) /

TypeError: must be str, not int"
Can you please advise?
Thank you

PR #223 (2016-07-22): add optional ivcurve calculation to singlediode
This PR adds an IV curve calculation feature to pvsystem.singlediode. 

As discussed in #83, the singlediode API remains unchanged for timeseries analysis: series in yields dataframe out. Scalar and array input still yield dict output (well, now an OrderedDict). 

Setting the new ivcurve_pnts keyword argument to a non-zero number will always yield an OrderedDict. The dict will also include the keys i and v. The values will be arrays with dimension one greater than the input data (1d arrays for scalar input, 2d arrays for array/series input).

Hop [... cut]