pvlib/pvlib-python #84: Tutorilal: system modeling

dacoex, 2015-08-20:
NOTE: no bug but clarification question.

Reading the tutorials is is unclear how inverter connections are defined and the power of a whole PV system of more than 1 string is calculated.

The notebook shows that 1 module is picked and then DC power calculated.
http://nbviewer.ipython.org/github/pvlib/pvlib-python/blob/master/docs/tutorials/tmy_to_power.ipynb#DC-power-using-SAPM

But the next step
http://nbviewer.ipython.org/github/pvlib/pvlib-python/blob/master/docs/tutorials/tmy_to_power.ipynb#AC-power-using-SAPM

uses a micro-inverter on module level.

How would the whole process be done with a different inverter, say a 30kW multi-string inverter?

And how would these then be added into a system of 500kW?

If it's roof-top, then multiplying by number of inverters would be possible but on a flat roof or ground mounted case, row-to-row spacing would need to be accounted for.

wholmgren, 2015-08-21:
Feel free to make PRs with improved tutorials.

dacoex, 2015-08-21:
Well, that was the purpose of my comment: I am not really sure how it is done with pvlib.

Although I could see the potential applications already.

wholmgren, 2015-08-25:
There is currently no standardized way to do something like that in pvlib. Would be nice to see development in that direction.

dacoex, 2015-08-25:
Ok, so it means that your application of pvlib is rather for forecasting. There, single system is not that important but rather an upscale get of the typical behaviour, right?

My focus would be more system analysis.

Back to the topic: so the 30kW inverter would be simulated with a, say a number of 138 modules of 260Wp totalling to about 36kW.

So in section 
http://nbviewer.ipython.org/github/pvlib/pvlib-python/blob/master/docs/tutorials/tmy_to_power.ipynb#AC-power-using-SAPM
One would have to manually add the strings to series and parallels, correct?

jforbess, 2015-08-25:
Yes, I have put together a dict that I use that expands the systemdef, and
perhaps it is time to wrap it into the systemdef dict, because it is pretty
unhelpful to define how many modules you have if you don't define which
kind of modules they are, or what kind of inverter you are using.

One more item on the todo list.

jforbess, 2015-08-25:
My dict isn't anywhere to be found in the pvlib code, in case it wasn't
clear.

wholmgren, 2015-08-26:
Realistic system level modeling in pvlib needs expert help. I don't have the time or the expertise to do it right. There is discussion in #17 and #39 that is relevant.

And yes, upscaling small, idealized systems is fine for forecasting applications.

dacoex, 2015-08-26:
@jforbess 
Would it be possible to see a demo of your dict in a notebook?

>Realistic system level modeling in pvlib needs expert help. I don't have the time or the expertise to do it right.
Sure. But what I like about the lib is that it can help to understand inner workings of the system well.
So we could add a joint notebook of all interested in that part.

An example might be what statsmodels team did (https://github.com/statsmodels/statsmodels/). They'd compared their models & results with existing and established solutions:
https://groups.google.com/forum/#!searchin/pystatsmodels/stata

This can be done for pvlib with SAM:
https://sam.nrel.gov/sites/sam.nrel.gov/files/sam-webinars-2013-sdk.pdf
https://sam.nrel.gov/content/lk-vs-python

>And yes, upscaling small, idealized systems is fine for forecasting applications.
Thanks for this confirmation. This helps to understand the motivation for the development focus.

jforbess, 2015-08-26:
I would be happy to share, once I clean it up from work I have done for a
client. Next week might be a good time for me to do that. Please feel free
to ping me again if it hasn't happened by Sept 8.

wholmgren, 2015-09-08:
@jforbess looking forward to seeing your ideas

dacoex, 2015-09-22:
see also:
https://github.com/oemof/feedinlib

dacoex, 2015-11-03:
@jforbess 
The tutorial looks good:
http://nbviewer.ipython.org/github/jforbess/pvlib-python/blob/Issue84/docs/tutorials/system_loss_modeling.ipynb

I wonder if this effort is doing the same as here in:
https://github.com/oemof/feedinlib/blob/master/example/feedinlib_example.py#L134

For reference:
https://github.com/pvlib/pvlib-python/pull/88
https://github.com/jforbess/pvlib-python/blob/Issue84/docs/tutorials/system_loss_modeling.ipynb