pvlib/pvlib-python #85: Inverter model supporting temperature derate?

jforbess, 2015-08-26:
The snlinverter doesn't support an adjustment for temperature derating. My experiments with implementing it after the fact show a gap due to the way the coefficients define the curve to the maximum output, Pac0, which changes based on temperature. 

I saw that in issue #2 there was a request for a new inverter model. Does that model support temperature derate, or should I work on defining an snlinverter_with_temperature_derate? I think it should be a separate function because the snl inverter model definition doesn't seem to include this concept, and we probably want to keep our models consistent with their references?

wholmgren, 2015-08-26:
I have wanted this as well, though I have no expertise in the area. I agree that it should be separate from the snl inverter implementation.

wholmgren, 2016-06-28:
xref sandialabs/MATLAB_PV_LIB#6