pvlib/pvlib-python #86: add forecasting tools to pvlib

wholmgren, 2015-09-08:
Hello fellow pvlib-ers,

I've found some funding to develop open source solar power forecasting tools based on pvlib python and US weather models. EPRI and Southern Company will fund a U. Arizona grad student, Derek Groenendyk, (@MoonRaker) to do the majority of the work this fall semester, though I will work closely with Derek on the project and I will maintain it in the future. The hope is that the pvlib-based forecasts can be reproducible and documentable benchmarks for the solar power forecasting community.

My plan is to add a new ``forecasting.py`` module to pvlib python, though I think we should have a conversation about if it's better to add it as a separate project under the pvlib organization. If the community is opposed to this too, then we will host it on our UA-REN organization account.

Unidata is putting together a nice project called [Siphon](https://github.com/Unidata/siphon) that makes it relatively easy to access data from a number of US weather models (though some cover more than just the US). I experimented with Siphon in [this notebook](http://nbviewer.ipython.org/github/wholmgren/siphon/blob/solar_testing/examples/notebooks/solar_testing.ipynb) and concluded that this project was very feasible.

One thing that we will need to do is come to some sort of agreement on how to spec out PV systems in pvlib. This has been discussed in a number of other issues in pvlib including #84 #17. So, I think this project should benefit pvlib users even if they're not interested in solar power forecasts.

Thanks in advance for your input on this. 

I'll ping a few people just to make sure that they see this... @bmu @Calama-Consulting @jforbess

jforbess, 2015-09-09:
Wow. Open source forecasting sounds great. I believe it will be a definite
advantage to some pvlib users, so I view it as reasonable to include in the
pvlib-python umbrella, but others may have a different view.

I really need to dust off the pvsystem config work. Friday?

bmu, 2015-09-18:
Would be nice to have something included in pvlib. However, if this project is mostly focussed on collecting meteorological data for the forecast (pv modelling may be only a small part of it) it may be a good idea to add it as a separate project under the pvlib organisation.

wholmgren, 2016-07-28:
closed by #180.

closed 2016-07-28 by wholmgren