pvlib/pvlib-python #87: test against python 3.5, pandas 0.17

wholmgren, 2015-09-14:
Python 3.5 was released yesterday (Sep 14, 2015) and Pandas 0.17 is scheduled to be released sometime this week. I'll add them to the travis build matrix later this week. I don't expect any problems.

wholmgren, 2015-10-10:
Pandas 0.17 is now out, so I'm back to trying to make this work. There are no new real code compatibility problems introduced by Pandas 0.17 and Python 3.5, but there are new packaging problems. 

* The Linux/Python3/SciPy issue #72 continues to exist on the most recent versions of those packages. On Python 3.4, you could hack around it by downgrading the numpy and scipy builds, but this is not easily possible on Python 3.5 for two reasons. 1) Continuum's Pandas 0.17 is only built against the new numpy 1.10, and 2) Continuum's SciPy 0.14 is not available for Python 3.5.
* Pandas 0.17 is not available for Python 3.3. I suggest that we remove 3.3 from the supported distributions. I don't know of any reason why one would stick with 3.3 over 3.4 or 3.5, and if you're using Python 3 then you're probably a person that likes to keep things up to date.

I just downloaded scipy 0.17dev and I'm still having problems, so this linux issue isn't going to magically disappear anytime soon.

jforbess, 2015-10-11:
Seems reasonable to drop python 3.3, given the development level of pvlib,
and the ability to use virtual environments if absolutely necessary.

(Speaking as someone who prefers to not update until forced, but recognizes
the value of being pushed to upgrade somewhat regularly.)

PR #90 (2015-10-27): Python 3.5, reconfigure Travis
This PR primarily does two things:
* Allows pvlib to be installed on Python 3.5
* Reconfigures the TravisCI set up to work around Continuum's crappy scipy/linux/python3 package.

I originally thought that there was an issue with scipy/linux/python3, but it turns out that it's actually a problem with the version of gcc that Continuum uses to build scipy. For more, see ContinuumIO/anaconda-issues#425. The Readme text has been updated accordingly. 

The Travis build matrix was getting to be too big and take too long, so this PR only tests the minimum pandas/numpy ve [... cut]
closed 2015-10-27 by wholmgren