pvlib/pvlib-python #99: doc: add tutorial to convert input ts to tilted

dacoex, 2015-11-11:
Say you receive a ts that includes 

*ghi*

or 

*ghi, dhi, dni*

would be nice to demonstrate how to convert the values to poa_global.

It would be complementary to the  existing or a variation thereof:
https://github.com/jforbess/pvlib-python/blob/master/docs/tutorials/tmy_and_diffuse_irrad_models.ipynb

wholmgren, 2015-11-11:
I don't understand your proposal. Plane of array calculations are documented in the tutorial you linked to, the [irradiance.ipynb](https://github.com/pvlib/pvlib-python/blob/master/docs/tutorials/irradiance.ipynb) notebook, and the [documentation](http://pvlib-python.readthedocs.org/en/latest/pvlib.html#pvlib.irradiance.total_irrad).

Feel free to create a pull request with improvements.

closed 2015-11-11 by wholmgren

dacoex, 2015-11-11:
That's what I plan.

But you can keep it closed since it's not a bug...