pvlib/pvlib-python #102: doc: add variables

dacoex, 2015-11-24:
Hello,

anything against adding
https://github.com/pvlib/pvlib-python/wiki/Variables-and-style-rules
to the sphinx docs?

this way, I have them with me also when offline.

wholmgren, 2015-11-25:
Sure, that makes sense now that the names have stabilized. Can you make a pr?

dacoex, 2015-11-25:
sure.

I suggest to use 
http://docutils.sourceforge.net/docs/ref/rst/directives.html#csv-table

Do you agree?

wholmgren, 2015-11-30:
Why do you prefer this to the standard sphinx table format? I think the sphinx format is easier to read in the doc source code.

dacoex, 2015-11-30:
Was fastest to implement.
Additionally, the csv values can be reused, e.g. to assign variables or translate variables into descriptions.

wholmgren, 2015-11-30:
Ok, well I am confused about the status since you closed your PR and deleted your branch. Please provide a link to rtd's rendering your branch when you (re)open your PR.

dacoex, 2015-12-01:
I am definitly not a git pro. But I tried to pack my 3 commits together. So maybe github closed it through the commands.
The link is here:
https://github.com/dacoex/pvlib-python/pull/2
Thanks.

dacoex, 2016-01-15:
Shall we also compare the parameter names with [IEC 61724](https://law.resource.org/pub/in/bis/S05/is.iec.61724.1998.pdf)

wholmgren, 2016-01-15:
You'll have to be more specific for me to understand what you're really proposing.

dacoex, 2016-01-18:
See:
https://github.com/pvlib/pvlib-python/commit/e6dea98a855bb93e325ef95c632b407d01c6ceed#diff-01c51400deeacb229d692d6a21e7af79R17

wholmgren, 2016-01-18:
I'm ok with adding a symbol column, but I don't think we should tie it to the IEC document. I would default to the PVPMC symbols.

dacoex, 2016-01-18:
> I would default to the PVPMC symbols.

sure, the other one is just for reference.

But it's good in the case one performs performance evaluation with usually bases on the IEC or the contractual formulas.

wholmgren, 2016-01-18:
My preference is for a simple table that is focused on the pvlib-python standards. A handful of links to documents such as the IEC's and PVPMC website could be useful below the table.

dacoex, 2016-01-18:
mmh.

wholmgren, 2016-01-18:
I don't know what that means.

dacoex, 2016-01-18:
Let's start with your proposal. 
Ideally, I'd love to have variables, units and refernce to standard side-by-side.

dacoex, 2016-01-19:
Comment https://github.com/pvlib/pvlib-python/issues/102#issuecomment-172568309 implemented in 
https://github.com/pvlib/pvlib-python/pull/103

2 items I find strange:

* RTD stylesheet / template does not support bulleted lists, see the test list below.
* RTD stylesheet / template does not support line wrapping in CSV table directive.

I tested these 2 items with a fresh sphinx project and there it worked.

Looking forward to your feedback to be able to close this issue ;-)

dacoex, 2016-01-25:
Clsong since the PR #103 was merged.

closed 2016-01-25 by dacoex