pvlib/pvlib-python #105: harmonise function output variables

dacoex, 2016-01-05:
Reference:
https://github.com/pvlib/pvlib-python/wiki/Variables-and-style-rules

on the wiki is days `poa_global` but `´pvlib.irradiance.total_irrad` produces the column `total`

dacoex, 2016-01-05:
So it woudl look like:

```
df_total_irrad.rename(columns = {'beam' : 'poa_direct',
                                 'ground' : 'poa_ground_diffuse',
                                 'sky' : 'poa_sky_diffuse',
                                 'total' : 'poa_global'},
                                 inplace = True)
```

wholmgren, 2016-01-05:
Yes, I think they should be changed.

dacoex, 2016-01-05:
assign it to me, then.

wholmgren, 2016-01-05:
Are you intending to use this issue for multiple functions or just the total_irrad function? Either is fine, I just want the issue title to be accurate.

In any case, I believe that you have to assign it to yourself, though I don't use this GitHub feature very often.

dacoex, 2016-01-05:
>Are you intending to use this issue for multiple functions or just the total_irrad function?

So far only this one. I did not encounter more similar items.

>In any case, I believe that you have to assign it to yourself, though I don't use this GitHub feature very often.

No, it would be you here:
https://help.github.com/articles/assigning-issues-and-pull-requests-to-other-github-users/

It would help to keep track of my tasks.

wholmgren, 2016-01-05:
The only organization members are the people that have commit rights. I
don't want to complicate it at this time.

closed 2016-03-09 by wholmgren