pvlib/pvlib-python #107: best pratice: impact of timezone shifts to data

dacoex, 2016-01-05:
Fellow pvlib users,
I would like to ask how you handle timezone (TZ) shifts of data supplied by the provider in UTC.
When the `tz_convert` is applied, one ends up with incomplete days at beginning and end of the series.

Let's say I have a timeseries for the whole of 2014 in hourly data.

```
import numpy as np
import pandas as pd

days =365 * 24 * 60

rng = pd.date_range('1/1/2014', periods=days, freq='T')
ts = pd.Series(np.random.randint(low=200, high=1000, size=len(rng)), index=rng)

tz_dataset = 'UTC'
tz_target = 'Atlantic/Bermuda'

ts_utc = ts.tz_localize(tz_dataset)

ts_utc.index.summary()
Out[345]: u'DatetimeIndex: 525600 entries, 2014-01-01 00:00:00+00:00 to 2014-12-31 23:59:00+00:00\nFreq: T'
```

Then I convert it to local
```
ts_tz = ts_utc.tz_convert(tz_target)
ts_tz = ts_tz.asfreq(ts_tz.index.inferred_freq)

```
naturally, the values get moved into the previous day:
```
ts_tz.index.summary()
Out[344]: u'DatetimeIndex: 525600 entries, 2013-12-31 20:00:00-04:00 to 2014-12-31 19:59:00-04:00\nFreq: T'
```
This is then significant when one wants to have complete days, retrieve extact time matching the measured data or aggregate to monthly data.

Any hints or help is appreciated. Thanks in advance.

Note:
Although being more a pandas issue, I think this is more prelevant to climate / solar data in this context.

wholmgren, 2016-01-05:
Use the ``tz`` keyword argument.

http://pandas.pydata.org/pandas-docs/version/0.17.1/generated/pandas.date_range.html?highlight=date_range#pandas.date_range

wholmgren, 2016-01-05:
Or specify the timezone the as part of your start argument.

dacoex, 2016-01-06:
>Use the tz keyword argument.

My use case is that I read the data from a CSV file which was provided from a data supplier. 
This way, the `ìndex` and `freq` are inferred from the CSV file columns.

>Or specify the timezone the as part of your start argument.

How is this different from using the `tz` keyword argument?

dacoex, 2016-01-06:
Here is the updated example code which works as expected. 

```
import datetime as dt

import numpy as np
import pandas as pd
import pvlib as pv
import pytz

#XXX: sample data which is usually read by pd.read_csv
#HINT: in this case datetimes & frequency are inferred from the file
days =365 * 24 * 60
rng = pd.date_range('1/1/2014', periods=days, freq='T')
ts = pd.Series(np.random.randint(low=200, high=800, size=len(rng)), index=rng)

#XXX: time zone definition
tz_dataset = pytz.timezone('UTC')
tz_target = pytz.timezone('Atlantic/Bermuda')
utc_offset = tz_target.utcoffset(dt.datetime.now())
utc_offset_h = utc_offset.total_seconds() / (60 * 60)

#xxx: time zone definition of the series
ts_utc = ts.tz_localize(tz_dataset)
ts_tz_shift = ts_utc.tz_convert(tz_target)
ts_tz_shift = ts_tz_shift.asfreq(ts_tz_shift.index.inferred_freq)

#XXX: mask the time series to the data period required
start_date = ts.index.to_datetime()[0]
end_date = ts.index.to_datetime()[-1]
ts_tz = pd.Series(data=ts_tz_shift,
                      index=pd.date_range(start=start_date, end=end_date,
                                          freq=ts.index.inferred_freq,
                                          tz=ts_tz_shift.index.tz),)

df_tz_shift = ts_tz_shift.to_frame()
df_tz = ts_tz_shift.to_frame()

# create pvlib Location object based on meta data

latitude_loc = 32.3,
longitude_loc =  -64.75
altitude_loc = 100
name_loc='Test'

surface_tilt = 30
surface_azimuth = 180

loc = pv.location.Location(latitude_loc,
                           longitude_loc,
                            tz=tz_target, altitude=altitude_loc,
                            name=name_loc)

solpos_min = pv.solarposition.get_solarposition(ts_tz.index, loc)
dni_extra_min = pv.irradiance.extraradiation(ts_tz.index)
airmass_min = pv.atmosphere.relativeairmass(solpos_min['apparent_zenith'])

irrad_min_poa = pv.irradiance.total_irrad(surface_tilt, surface_azimuth,
                          solpos_min['apparent_zenith'],
                          solpos_min['azimuth'],
                          dni=(ts_tz * 1.3),
                          ghi=(ts_tz),
                          dhi=(ts_tz * 0.4),
                          dni_extra=dni_extra_min,
                          airmass=airmass_min,
                          albedo=0.25,
                          surface_type=None,
                          model='perez',
                          model_perez='allsitescomposite1990')
```

Any suggestions and imporvements are welcome. 

Again, a `pandas` issue but the application of this library to typical data in the meteo domain is sometimes not easy because our data has phyical plausability limitations.
Most tutorials and questions relate to fincancial or random data where this is no much of a problem. 

So thanks again to @wholmgren for is patience ;-)

closed 2016-01-06 by dacoex

jforbess, 2016-01-06:
Thanks for sharing your answer. I have experienced that issue, but have not
worried about solving it yet. My standard problems with timezones are
usually due to DST being applied or taken away in a dataset. Or both.

Essentially this issue occurs when the dataset is provided with a tz that
is not the timezone it is actually in.

wholmgren, 2016-01-06:
I'm a bit confused. Can't you just call ``.tz_localize(timezone_that_the_data_was_recorded_in)`` on the data that you imported? Then you use your data's Series/DataFrame.index as an argument to the pvlib functions. I believe that this is what the TMY examples do.

A couple of other comments...

1. Best to specify dates using ISO 8601. You and I might disagree on the meaning of '1/2/2014'.
2. It's too bad that you have to create a ``Location`` object to do this. 
2. You'll get [syntax highlighting](https://help.github.com/articles/github-flavored-markdown/#syntax-highlighting) in github if you add the word python to the first line of your block quote.

> How is this different from using the tz keyword argument?

There are some subtle differences in their fancy representations...

```python
In [8]: pd.date_range('20140101', periods=3, freq='T', tz='US/Arizona')
Out[8]:
DatetimeIndex(['2014-01-01 00:00:00-07:00', '2014-01-01 00:01:00-07:00',
               '2014-01-01 00:02:00-07:00'],
              dtype='datetime64[ns, US/Arizona]', freq='T')

In [9]: pd.date_range('20140101-0700', periods=3, freq='T')
Out[9]:
DatetimeIndex(['2014-01-01 07:00:00', '2014-01-01 07:01:00',
               '2014-01-01 07:02:00'],
              dtype='datetime64[ns]', freq='T')

In [10]: pd.date_range('20140101T00:00:00-0700', periods=3, freq='T')
Out[10]:
DatetimeIndex(['2014-01-01 00:00:00-07:00', '2014-01-01 00:01:00-07:00',
               '2014-01-01 00:02:00-07:00'],
              dtype='datetime64[ns, tzoffset(None, -25200)]', freq='T')
```

However, they all have the same integer representations...

```python
In [6]: pd.date_range('20140101', periods=3, freq='T', tz='US/Arizona').astype(int)
Out[6]: array([1388559600000000000, 1388559660000000000, 1388559720000000000])

In [7]: pd.date_range('20140101-0700', periods=3, freq='T').astype(int)
Out[7]: array([1388559600000000000, 1388559660000000000, 1388559720000000000])

In [8]: pd.date_range('20140101T00:00:00-0700', periods=3, freq='T').astype(int)
Out[8]: array([1388559600000000000, 1388559660000000000, 1388559720000000000])
```

jforbess, 2016-01-06:
>    3. You'll get syntax highlighting

I think the key issue is that DaCoEx is given data starting on 20140101
0:00 +0:00, which is 20131231 20:00 -4:00, but wants to provide a report on
data only from 20140101 0:00 -4:00 through 20141231 23:59 -4:00, not
starting on 20131231 20:00 -4:00.

I can't quite tell if @wholmgren's solution addresses that, but I don't
think it does.

jforbess, 2016-01-06:
I guess it does simplify by applying a tz to the date_range filter, which
is simpler than DaCoEx's method.

dacoex, 2016-01-06:
>>I'm a bit confused. Can't you just call .tz_localize(timezone_that_the_data_was_recorded_in) on the >>data that you imported? Then you use your data's Series/DataFrame.index as an argument to the >>pvlib functions. I believe that this is what the TMY examples do.

My original explanation of the objective should have been better. But @jforbess fleshed it out in https://github.com/pvlib/pvlib-python/issues/107#issuecomment-169405157
The best would be to create a notebook showing the motivation etc. But I was in the middle for some processing. As you see,my issue is that I am not an every day pandas user. This is not because I dislike it but data processing is just a tiny part of the project work...

> Best to specify dates using ISO 8601. You and I might disagree on the meaning of '1/2/2014'.

OK. My example was a copy&paste from the pandas docs...

>   It's too bad that you have to create a Location object to do this.

That's the bit we wanted to change with the refractoring, right?

>    You'll get syntax highlighting in github if you add the word python to the first line of your block quote.

Changed above; I will use this in future.

>I guess it does simplify by applying a tz to the date_range filter, which

I do not see which part of my example this applies.

Are you referring to this section

```python
rng = pd.date_range('1/1/2014', periods=days, freq='T')
ts = pd.Series(np.random.randint(low=200, high=800, size=len(rng)), index=rng)
```

or this one
```python
ts_tz = pd.Series(data=ts_tz_shift,
                      index=pd.date_range(start=start_date, end=end_date,
                                          freq=ts.index.inferred_freq,
                                          tz=ts_tz_shift.index.tz),)
```

jforbess, 2016-01-11:
> @jforbess <https://github.com/jforbess> fleshed it out in #107 (comment)

Sorry for the delayed response. Yes, it was just simplifying your
date_range parameter.