pvlib/pvlib-python #117: docs: testing of notebooks

dacoex, 2016-01-25:
How are [notebook tutorials](https://github.com/pvlib/pvlib-python/tree/master/docs/tutorials) integrated in the testing framework?

wholmgren, 2016-01-25:
The tutorials are not integrated into the testing framework. Notebooks are a giant pain for git, testing, and refactoring code. I've seen a handful of projects that claim to make these things easier, but I haven't actually used them.

Most of the notebooks do have a note that states the versions that they have been tested with. I update the tutorials when I make or consider merging changes that I know will cause problems. That's certainly not a long term solution.

dacoex, 2016-01-25:
OK, so I will use doctest-able code listings for essential documentation.
The notebook may then only show certain further functionality or how to tie things together.

I like the way xarray created their examples: http://xarray.pydata.org/en/stable/examples.html

wholmgren, 2016-01-25:
Do we need to keep the issue open then?

I like the way that xarray does a lot of things.

dacoex, 2016-01-25:
>I like the way that xarray does a lot of things.

I was also thinking of check how pvlib orks with it. But prioritising is important, here...

closed 2016-01-25 by dacoex