pvlib/pvlib-python #122: Using cec-modules with sapm

uvchik, 2016-02-17:
As I understand it, it is not possible to use the `pvlib.pvsystem.sapm` function with the cec-modules-library.

Is this right?

If it is right, is it possible to write a hint somewhere in the documentation.

http://pvlib-python.readthedocs.org/en/latest/pvlib.html#pvlib.pvsystem.sapm

cwhanse, 2016-02-17:
The CEC library doesn’t contain the parameters that are needed for the SAPM.  It only has parameters for the CEC performance model.

NREL _might_ (I’m not sure) have added functionality to translate from CEC model parameters to SAPM parameters.  I know PVsyst has done this, but the translation is suspect in my opinion.  It works by computing modeled IV curves (using the PVsyst model), picking off Isc, Vmp/Imp and Voc, then fitting the SAPM equations to these _modeled_ points.  So you get a SAPM model which emulates the PVsyst model including any biases, rather than measured performance.

Cliff

http://pvlib-python.readthedocs.org/en/latest/pvlib.html#pvlib.pvsystem.sapm

wholmgren, 2016-02-17:
Sure, let's add a hint to the documentation. Feel free to make a PR, otherwise I'll get to it sometime in the next few weeks.

uvchik, 2016-02-18:
See PR #123.

wholmgren, 2016-02-18:
Closed by #123

closed 2016-02-18 by wholmgren