pvlib/pvlib-python #125: doc: how to get dhi & bni

dacoex, 2016-02-23:
`pvlib.clearsky.ineichen` produces 

* GHI 
* DHI 
* BNI

It is not yet documented, how the non-clear-sky diffuse horizontal and direct horizontal components are derived.
Say, 

I checked the tutorials. Either DHI is included in the TMY files or the DHI of clear-sky is used.

If my understanding is am correct, there is no procedure for deriving actual dhi, e.g. during overcast conditions, from a timerseries of GHI.
The same applies for bhi

Maybe a matrix wich function for converting which irradiance value from which start value using which function would be nice to have that overview.

wholmgren, 2016-02-23:
pvlib implements and documents several models that calculate DNI from GHI data:

http://pvlib-python.readthedocs.org/en/latest/pvlib.html?highlight=disc#pvlib.irradiance.disc

http://pvlib-python.readthedocs.org/en/latest/pvlib.html?highlight=dirint#pvlib.irradiance.dirint

dacoex, 2016-02-23:
Sure.
But I will still need to use the math for the DHI (GHI-dependant / non clear-sky)

wholmgren, 2016-02-23:
You mean  ``dhi = ghi - dni*cos(zenith)``? Do you want a function for this? Do you want to make a new documentation example? I'm not sure what you're asking for.

dacoex, 2016-02-24:
Yes, I will add this to my list. I will try to elaborate further on it.
I feel the description of the the irradiation module could need some overview at beginning and guidance.

wholmgren, 2016-02-24:
Based on my understanding, we should do one or both of the following:

1. Rename this issue "Add overview to irradiance module documentation".
2. Add your specific use case as an example to be addressed in issues #60 and/or #120.

Can you clarify or provide guidance?

wholmgren, 2019-05-03:
I think this was addressed by `irradiance.dni` and #692.

closed 2019-05-03 by wholmgren