pvlib/pvlib-python #140: Night time power issue for some microinverters

ejmiller2, 2016-03-21:
I am trying to use pvlib to compare some systems with microinverters versus string inverters. I have noticed that for many microinverters I am getting a positive AC power output when the DC power is 0. For example, using the inverter:
    `Enphase_Energy__M250_60_2LL_S2x___ZC____NA__208V_208V__CEC_2013_`

With 0V and 0W:
    `pvsystem.snlinverter(inverters.Enphase_Energy__M250_60_2LL_S2x___ZC____NA__208V_208V__CEC_2013_, 0, 0)`

I get:
    `1.7860471814922816`

I would expect to see a negative result with a magnitude of Pnt, which is what I see for most of the other string/central inverters I have tried. This is causing issues for my comparisons since my night time values for the microinverter are adding to the total power production, while the string and central inverters are being reduced. This can add up to quite a difference when accumulating over a year.

If I try the same module and same inverter in SAM, the "Power generated by system" is showing the negative Pnt value I expect for the inverter when the DC values are 0.

The code in pvsystem.py for this situation is
    `ac_power[ac_power < Pso] = - 1.0 * abs(Pnt)`

Shouldn't the comparison be between the DC power and Pso (p_dc < Pso)?

wholmgren, 2016-03-21:
I think your understanding is correct and the code should be changed. However, PVLIB Matlab's ``pvl_snlinverter.m`` specifies:

``ACPower(ACPower<Ps0)= -1.*abs(Pnt); % Inverter night tare losses``

so I would like to get some clarification/confirmation from @cwhanse or @jsstein. This is in at least PVLIB Matlab 1.2, 1.3 and 1.32dev (I was surprised to find that this is the current download available from PVPMC).

cwhanse, 2016-03-22:
That code should be ACPower(DCPower<Ps0) rather than ACPower(ACPower<Ps0).

Thanks for the note.

Cliff

wholmgren, 2016-03-22:
Thanks, Cliff. 

@ejmiller2 would you like to make a pull request with the fix? Either way is fine.

Let's plan on including this in a 0.3.1 release in a week or two.

ejmiller2, 2016-03-22:
@wholmgren,

Thanks for the quick turnaround. I'm currently running pvlib that I grabbed as a conda package. Once I figure out how to run from source, I may want to do a pull request. For now I'm ok at least knowing why there's a discrepancy in my results.

ejmiller2, 2016-03-23:
@wholmgren Hopefully I have done that correctly...

commit 566a1e13 (2016-04-12): Merge pull request #141 from ejmiller2/master

Resolve #140 snlinverter night tare power issue
commit c90b7d02 (2016-04-12): Resolve #140 snlinverter night tare power issue

Fixing #140 by changing the start power (Pso) comparison to use the DC input power (p_dc) instead of the calculated AC power (ac_power).
closed 2016-04-12 by wholmgren