pvlib/pvlib-python #143: create ModelChain(s) that implement very simple models

wholmgren, 2016-04-07:
It could be useful to create a ModelChain or two that implements some kind of extremely basic system modeling. Something like ``power = user_def_const * poa_global``. The PVUSA model could be useful too. You all probably have more experience with sort of thing than I do.

A couple of motivations: 
* The simple models would make it easier to quantify just how much value the more sophisticated models add in different situations.
* The simple models might be a more gentle introduction to the library.
* It's easier to fit simple models to data, or at least to get started fitting.

I'm not yet sure if/how PVSystem would need to be modified to support this kind of thing. Maybe some additional keyword arguments in its constructor. Maybe you can reuse the module and inverter parameters arguments.

wholmgren, 2016-04-07:
To be a little more concrete, the ``ModelChain`` refactor might look something like:

```python
class ModelChain():

    def prepare_inputs():
        self.solar_position = self.location.get_solarposition(self.times)

        self.airmass = self.location.get_airmass(
            solar_position=self.solar_position, model=self.airmass_model)

        if irradiance is None:
            irradiance = self.location.get_clearsky(
                self.solar_position.index, self.clearsky_model,
                zenith_data=self.solar_position['apparent_zenith'],
                airmass_data=self.airmass['airmass_absolute'])
        self.irradiance = irradiance

        self.total_irrad = self.system.get_irradiance(
            self.solar_position['apparent_zenith'],
            self.solar_position['azimuth'],
            self.irradiance['dni'],
            self.irradiance['ghi'],
            self.irradiance['dhi'],
            model=self.transposition_model)

        if weather is None:
            weather = {'wind_speed': 0, 'temp_air': 20}
        self.weather = weather

    def run_model():
        raise NotImplementedError('you must subclass MC and implement this method')

class SAPMChain(ModelChain):

    def run_model():
        self.prepare_inputs()
        # followed by all of the SAPM calls

class BasicChain(ModelChain):

    def run_model():
        self.prepare_inputs()
        self.dc = self.total_irrad['global'] * self.poa_global_to_dc_power_const
        self.ac = self.dc * self.dc_to_ac_const

class PVUSAChain(ModelChain):

    def run_model():
        self.prepare_inputs()
        # the pvusa eqn

```

dacoex, 2016-04-23:
hello again ;.)

I'd done this with the 2.x code but still quite experimental.

I would recommend to use:
PR instead of `poa_global_to_dc_power_const`. The user could take the PR from from other references (PVSyst, performance evaluation).

jforbess, 2016-04-23:
I hate to say it, but I recommend against PR for this const. Often PR is
calculated through to ac power (PVsyst), so it would be a mistake to assume
it's appropriate to apply to dc. I agree that the current name is awkward,
but I think that is the pvlib way, more or less.

Also, I think PR is the most overloaded term in solar, and have seen it
used in a number of ways that aren't the traditional kWh / kWdc / POA.
While PR had its uses for years, I'm hoping that it is soon to be
deprecated for most uses through the use of programs like pvlib.

cwhanse, 2016-04-25:
I agree with Jessica.  One common issue with PR is that there are several definitions in current use.

Perhaps we can agree on the simple model and that would help clarify things.

My proposal would be PVWatts:

Pac = POA/1000 x Pdc(nameplate) x efficiency x (1 + gammaPmp (Tc – 25))

We need three constants: Pdc(nameplate), efficiency (PVWatts uses 0.81)  and gammaTmp (-0.004/C is typical)

Cliff

wholmgren, 2016-04-25:
+1 for PVWatts. Might be a good project for someone at the EPRI workshop.

wholmgren, 2016-07-28:
closed by #194.

closed 2016-07-28 by wholmgren