pvlib/pvlib-python #144: inconsistency between PVlib and PVsyst when calculating rotated plane azimuth

mayudong, 2016-04-07:
When I was comparing the stimulative results of PVlib with PVsyst, I noticed that for PV panels with same configurations, the calculated rotated plane azimuth computed by pvlib.tracking.singleaxis(..)
is different from that of PVsyst. 

I did find out where the problem is, and it is related to the coordinator transformation.

Can I create a pull request to submit my fix?

wholmgren, 2016-04-07:
@mayudong yes, that would be great.

tafadzwalgurupira, 2016-04-11:
I also noticed different results for my POA irradiance results when I used the tutorial(tmy_to_power) and compared the results with PVSyst. Can anyone help me figure out where I am going wrong and fix it?

wholmgren, 2016-04-11:
@tafadzwalgurupira, I believe that @mayudong is discussing a potential issue with the single axis tracker code, whereas you are discussing an issue with some part of the plane of array calculations. So, I'd prefer that you make a new issue with more details to discuss the discrepancy between the tmy_to_power tutorial and your pvsyst results. This helps keep discussions focused and on topic. Thanks.

cwhanse, 2016-04-11:
@mayudong  could you post a summary of your fix?  If there's a code error, it's likely also present in the Matlab toolbox and I'd like to fix it there as well.  Thanks.

mayudong, 2016-04-11:
Sure. I have summarized the issue and potential fix in the attached file.
Please double check and confirm the issue.
Best
[NEX-PVSystemModelComparison-110416-1221-2.pdf](https://github.com/pvlib/pvlib-python/files/213735/NEX-PVSystemModelComparison-110416-1221-2.pdf)

wholmgren, 2016-04-11:
I tried to simply this part of the matlab code and it's possible that I introduced an error. I'll look at it more carefully.

mayudong, 2016-04-11:
Actually the results calculated using Matlab are different from PVsyst too.
I will create a pull request sometime this week

wholmgren, 2016-04-12:
I looked at the surface azimuth results more carefully and I agree that they are not correct in pvlib python for non-trivial cases. Example 3 of the python [single axis tutorial](http://nbviewer.jupyter.org/github/pvlib/pvlib-python/blob/v0.3.0/docs/tutorials/tracking.ipynb#Example-3:) and the matlab version's [documentation](https://pvpmc.sandia.gov/PVLIB_Matlab_Help/html/pvl_singleaxis_help.html) simulates a tracker with 20 deg surface tilt. I remade the matlab figure to include the surface azimuth...

[image]
[image]

I'll look at the proposed fix in #145  next.

wholmgren, 2016-04-13:
Closed by #145

closed 2016-04-13 by wholmgren