pvlib/pvlib-python #146: Inconsistency between PVlib and PVsyst when calculating POA irradiance

tafadzwalgurupira, 2016-04-12:
I used the tutorial (tmy_to_power) to calculate the POA irradiance and compared them to a PVSyst simulation for the same conditions and settings and found the results varying significantly. 

screenshots of the simulations are here: https://www.dropbox.com/sh/3qtkozqde15i38c/AAA9_fERkvkxlW8gGyG3zJJja?dl=0

cwhanse, 2016-04-12:
If you are using PVsyst v6, the default transposition model is Perez, not Hay and Davies.  The PVLib tutorial uses Hay and Davies.

Cliff

tafadzwalgurupira, 2016-04-13:
@cwhanse , thanks for the suggestion but I had already checked it and that is not the case. I used Hay and Davies as prescribed and the results are only consistent when surface tilt is 0

wholmgren, 2016-04-13:
you're probably going to need to post your code if we're going to get to the bottom of this.

tafadzwalgurupira, 2016-04-22:
https://www.dropbox.com/s/9stmuuaavxdq7dp/tmy_to_power-Copy2.ipynb?dl=0

wholmgren, 2016-04-22:
Ok, focusing on July 5...

Here's the TMY data

[image]

The DNI reaches 800 W/m2 hr.

Next, the apparent solar zenith and the AOI for a surface_tilt=30 panel.

[image]

Finally, POA irradiances for flat and 30 degree panels.

[image]

[image]

These pvlib results make sense to me. Here's your "PVSystPOA_surface_tilt_30" plot...

[image]

Maybe something is mislabeled or we're using different conventions, but this does not make sense to me.

jforbess, 2016-04-22:
I loaded Sand Point into PVsyst 6.37 and got this graph:

Is it possible you are using a non due south azimuth in PVsyst? Note that
PVsyst thinks that 0 is due south and 180 is due north. Super annoying.

​

wholmgren, 2016-04-22:
@jforbess I think you need to use the GitHub website to include graphics in a comment.

jforbess, 2016-04-22:
Ah, good to know. 
I loaded Sand Point into PVsyst 6.37 and got this graph:
[image]

wholmgren, 2016-04-22:
@jforbess thanks! It looks reasonably consistent to me given that the pvlib tutorial's treatment of time for the hourly average modeling is going to cause some errors. I'm not sure how these PVSyst models are treating the time averaging.

jforbess, 2016-04-22:
Yes, I think so too.

tafadzwalgurupira, 2016-04-25:
Thanks for all your help everyone. The problem looks like it was just as @jforbess suspected: PVSyst's treatment of 0 and 180 degrees as due north and due south, respectively.

closed 2016-04-25 by wholmgren