pvlib/pvlib-python #152: sam files are not available

mikofski, 2016-04-29:
https://sam.nrel.gov/sites/sam.nrel.gov/files/sam-library-cec-inverters-2015-6-30.csv

and

https://sam.nrel.gov/sites/sam.nrel.gov/files/sam-library-sandia-modules-2015-6-30.csv

return HTTP/1.1 404 - Not Found

yotf, 2016-04-29:
yes, they seem to have change url.

Sandia module can now be found at https://sam.nrel.gov/sites/default/files/sam-library-sandia-modules-2015-6-30.csv

CEC inverters at https://sam.nrel.gov/sites/default/files/sam-library-cec-inverters-2015-6-30.csv

wholmgren, 2016-04-29:
That's unfortunate. We'll need to fix this before the EPRI modeling workshop, so that probably means a 0.3.2 release early next week. Maybe we should take care of #52 at the same time.

mikofski, 2016-04-29:
I'm working on a pull request to just get the data from the pvfree api. And also updating pvfree with modules from both Sandia and cec.

I think this is a better solution than static links or files.

Sunpower will host pvfree

wholmgren, 2016-04-30:
@mikofski I'm interested in your PR, just trying to get things working again as quickly as possible. Feel free to make a new issue for the pvfree api, of course.

closed 2016-04-30 by wholmgren

uvchik, 2016-05-02:
@mikofski : A PR to get the data from pvfree is really a good idea :+1:

PR #153 (2016-04-30): update sam file base url
Closes #152.