pvlib/pvlib-python #160: port pvl_adrinverter

wholmgren, 2016-05-06:
This is a relatively simple function, so it should be easier to port.

Including the database can be as simple as adding it to ``pvlib/data`` and adding it to the ``manifest``. Ideally we'd convert the database to something other than .mat (.csv or .h5), but that can also be done later.

This was previously mentioned in #2, but I want to highlight it here. 

cc @adriesse

adriesse, 2016-05-08:
A bit busy with other things at the
      moment, but feel free to ask questions about this.  
      
      Have a great workshop everyone!
      
      Anton
      
      
      On 2016-05-07 0:31, Will Holmgren wrote:
    
    
      This is a relatively simple function, so it should be easier to
        port.
      Including the database can be as simple as adding it to pvlib/data
        and adding it to the manifest. Ideally we'd
        convert the database to something other than .mat (.csv or .h5),
        but that can also be done later.
      This was previously mentioned in #2,
        but I want to highlight it here. 
      cc @adriesse
      —
          it on GitHub

VolkrB, 2016-12-21:
Hi all, 
I took the liberty of taking the Driesse inverter database .mat file and converting it to a pd.df and then to .csv. 

Question: I noticed that some of the inverters dont have model names, but do have the manufacturer and I can't directly check the .mat file because I don't have matlab installed here. (I converted the .mat with scipy)

Example: I found that there is some data where the manufacturer, Dow Chemical, doesn't have a model name. Could someone confirm this.

I want to be sure that my conversion went smoothly before I pull request

Thanks!

adriesse, 2016-12-21:
Thanks for the initiative!  I see four inverters from Dow Chemical without model descriptions in my source files.

VolkrB, 2016-12-21:
Great! That matches up with what I have. Going to see if I can get this csv to be compatible with pvsystem.retrieve_sam() and then port the pvl_adrinverter function. Let me know if you guys have any thoughts or concerns about this

adriesse, 2016-12-22:
I'll support your efforts.

wholmgren, 2016-12-23:
Awesome. In the short term, I agree that it makes sense to add a csv file to the ``data`` folder and ``name == 'adrinverter'`` option to the ``retrieve_sam`` function. I suppose someone might object to putting this in a SAM reader, but I don't think it's a big deal. In the long term, we should consider moving this kind of functionality to the new iotools subpackage (#261 and links therein).

There's also a stub method for the adrinverter in ``ModelChain`` that raises ``NotImplementedError``. Depending on your time and interest, you could implement that wrapper too. It looks to me like it would be pretty straightforward if you follow the ``ModelChain.snlinverter`` example, but don't feel bad leaving it for future work.

VolkrB, 2017-01-04:
I noticed *.csv in the .gitignore. Just checking if it's ok for me to -f commit the csv i've built out of the .mat file or should this be handled some other way?

wholmgren, 2017-01-04:
It's fine to force add it.

wholmgren, 2017-02-14:
closed by #288

closed 2017-02-14 by wholmgren