pvlib/pvlib-python #163: Adding New Single Diode Model For CdTe

crimmj, 2016-05-10:
Would like to add a new single diode model that includes recombination current for CdTe modules. There are some additional parameters in the model. Any thoughts on the correct structured way to do this?  Would this function or functions be added to pvsystem.py?

wholmgren, 2016-05-11:
Sounds useful to me. The existing functions that use module parameters will work with anything that supports dictionary-like lookups. Most of the time, people will be passing in Series objects from the SAM DataFrames, but plain dictionaries should work too. I would say that we should keep things simple and use a similar interface for new functions.

cwhanse, 2018-06-11:
This request essentially asks to implement the PVsyst model for a-Si, which includes a new term to account for recombination loss in the intrinsic layer (see [http://files.pvsyst.com/help/thinfilm_recombinationloss.htm](url)

PVsyst provides a default value for the single new parameter (di^2/mu teff = 1.4V).

Implementing this new model depends on having `calcparams_PVsyst` see #470 and will require a new function to solve the diode equation with the recombination current term.

mikofski, 2018-06-12:
I have this model already, I can try to add this

cwhanse, 2018-06-12:
@mikofski happy to have a volunteer!  Are you thinking to implement the PVsyst model (addressing both #470 and #163) or only the variation of `singlediode` to solve the current-voltage equation?  If you need help with #470 I can do the `calcparams_pvsyst` component.

mikofski, 2018-06-17:
If the recombination loss current is already included in the MATLAB version, then I guess we should just pretty that over here. @cwhanse are you planning to do that or are you looking for volunteers?

cwhanse, 2018-06-17:
I have some MATLAB code which isn't in PVLib.  I didn't find an elegant solution, it's brute force getting the current at each voltage.

mikofski, 2018-06-17:
Okay let's collaborate then. I use the same explicit approach for pvsyst as I do for and SDM. Then either interpolate or Newton method or bisection for specific points or the max p. I'll try to put something up this week. I think this counts as work don't you? On a related note, I added [an issue on SAM SSC on the recombination current for this film](https://github.com/NREL/ssc/issues/155). I think it should be added to the CEC and DeSoto models, the J. Merten, Mermoud, and Lejeune seem to indicate the an error would be present with this film for any SDM near Voc missing this term. Do you have an opinion? Maybe we can collaborate with FirstSolar.

cwhanse, 2018-06-18:
I think I need to be clear on the plan here.  This issue asks for a new function (counterpart to `pvsystem.singlediode` that solves

I = IL - IL*mu/[Vbi - (V+I*Rs)] - I0*[exp((V+I*Rs)/(nNsVth))-1] - (V + I*Rs)/Rsh

returning the IV curve. The second term on the right is the recombination current.

#470 asks for `calcparams_PVsyst` which is related, but basically independent, because the parameters unique to the recombination term (mu, Vbi) are constants.

We can work together to implement `singlediode_recomb` or whatever name we settle on.  I'll email my Matlab function to you.

mikofski, 2018-06-18:
Yes, I'm clear. What I was asking about was that since the only difference between the existing single diode model (SDM) and a model that included the [thin film recombination loss current](http://files.pvsyst.com/help/thinfilm_recombinationloss.htm) is the term:

    I_rec = I_L * d2_mu_tau / (V_bi - V_diode)

then:

* does it really make sense to create a separate `sdm_thin_film` method that is nearly identical to the existing SDM method?
* would it be possible to just include `I_rec = I_L * d2_mu_tau / (V_bi - V_diode)` into the existing `pvsystem.singlediode` and just turn it on as needed?

As I said previously, the solution for either is exactly the same: specify a bunch of diode voltages, solve for cell current then back calculate the cell voltage. For specific voltages and max power, then you can either interpolate, use Newton, or use a bisection search method. (Maybe Lambert-W also works but I'm not as familiar with that formulation?)

I understand that `calcparams_pvsyst` merely generates the 5 parameters that SDM uses since PVSyst has different auxiliary equations, so I understand that's the focus of #470.

So, as I asked above, for #163 we could either implement a new SDM method for thin film with recombination losses or just combine it with the existing SDM.

Regarding:

>I think it should be added to the CEC and DeSoto models, the J. Merten, Mermoud, and Lejeune seem to indicate that an error would be present with thin film for any SDM near Voc missing this term. Do you have an opinion? Maybe we can collaborate with FirstSolar.

Sorry to be so vague with that comment. What I meant was the question I asked above: shouldn't we just include the recombination loss current into the existing `pvsystem.singlediode` such that it would be available to any `calcparam_*()` method, not just PVSyst? I was thinking that by collaborating with FirstSolar we could determine whether or not the existing SDM without the recombination loss current exhibits the errors that J. Merten, Mermoud, and Lejeune observed.

adriesse, 2018-06-19:
I think it makes sense to incorporate this feature into the SDM functions.  The various models can then use the feature as needed. Maybe it even opens up some unexplored territory.

cwhanse, 2018-06-19:
I'm tracking with you now @mikofski  If we use the diode voltage method to solve the IV curve this new term is not a challenge. But `singlediode` and its helpers, `i_from_v` and `v_from_i` are using the LambertW method which doesn't extend when the recombination current is added to the single diode equation.

We had a long conversation about changing to the diode voltage approach #409 which hasn't been merged.  We need to resolve that before moving ahead with this issue. 

Since it has been a while, @mikofski can you remind us what issues remain with #409?

mikofski, 2018-06-21:
I've started trying to update #409, there's a list of TODO[s] that @wholmgren left for me to check off like fixing docstrings, better names (newton vs. "fast", etc.) and moving vendorized `scipy.optimized.newton` to `_tools` since it won't be merged until November (if ever :frown:)

mikofski, 2018-06-28:
TODO: see [this comment](../pull/409#issuecomment-401040515)

mikofski, 2018-07-11:
#### TL;DR:
Should we always limit IV curves for CdTe (and a:Si?) to the 1st quadrant only, or to a voltage in the 4th quadrant that is very close to open circuit voltage but much less than the intrinsic voltage?

#### More Discussion:

@cwhanse as voltage approaches the builtin voltage, the expression for the recombination loss current asymptotes to infinity:

    diode_voltage = voltage + current * series_resistance
    Irecombination = photocurrent * d2mutau / (intrinsic_voltage - diode_voltage)

@jdnewmil has suggested gradually turning off the recombination somewhere between open circuit (`V_oc`) and the intrinsic voltage of the thin layer, which should always be greater than `V_oc`.

There is nothing in [Meromoud's _Performance Assessment ..._](http://www.pvsyst.com/images/papers/thf_article_valence_2010.pdf) that suggests what to do.

Do you have any suggestions?

First to be clear, the recombination loss error, does not affect all thin films. For example, CIS/CIGS are **not** affected. Merten _et al_ and Mermoud _et al_ only observed the recombination loss for amorphous silicon (a:Si) and cadmium telluride (CdTe).

My idea would be to just limit IV curves for CdTe (and a:Si?) to the 1st quadrant only. This avoids the problem. There's no reason to go into the 4th quadrant for thin film because they typically don't have bypass diodes and therefore will fail badly if there is any partial non-uniform shade, _ie:_ only row-to-row shade that effects the entire array (except the front row) evenly. Also most CdTe (and a:Si?) are typically only installed on trackers, and are only shading when the sun is low in the sky since most  CdTe (and a:Si?) do not backtrack. Since the trackers are aligned N-S, the sun is nearly due east or west during sunrise/sunset, and the irradiance is very low. String voltage mismatch might be an issue at sunrise/sunset because of edge effects since the sun is not exactly due east or west, but the voltages should all be well below the intrinsic voltage due to the low irradiance, and voltage mismatch would be small because the sun is nearly due east or west.

Thanks for your comments.

cwhanse, 2018-07-11:
Yes, 1st quadrant only is fine from my point of view.

jdnewmil, 2018-07-12:
Really depends on what you plan to do with it. If you plan to study voltage mismatch then individual curves can operate in the fourth quadrant even while the overall array may not. My conclusion was that doing one quadrant only was unacceptable for my purposes.

On July 11, 2018 10:22:04 AM PDT, Cliff Hansen <notifications@github.com> wrote:
>Yes, 1st quadrant only is fine from my point of view.

-- 
Sent from my phone. Please excuse my brevity.

cwhanse, 2018-07-12:
@jdnewmil your use case is of interest to me also.  In this case I worry about applying the model at high external voltage. The way I read the reference is that this particular model for the recombination current loss derives from some assumptions that are valid at low external voltage, but questionable as the system moves towards Voc. For this reason I'm not eager to spend effort to make this calculation reliable in the 4th quadrant. I think we'd be better served looking for a modeling approach which is justified at high external voltages.

mikofski, 2018-07-13:
### Proposed changes as discussed in #409 
* on [`pvlib.singlediode_methods.bishop88` lines 103-104](../blob/master/pvlib/singlediode_methods.py#L103-L104) append
    ```python
    -photocurrent * d2mutau / (Vbi - diode_voltage/Ns)
    ```
    where `Ns` is the number of cells in series, `d2mutau` is a parameter, and `Vbi` is the intrinsic thin-film "built-in" voltage per junction.
* on [`pvlib.singlediode_methods.bishop88` line 109](../blob/master/pvlib/singlediode_methods.py#L109) append
    ```python
    -photocurrent * d2mutau / (Vbi - diode_voltage/Ns)**2
    ```
* on [`pvlib.singlediode_methods.bishop88` line 114](../blob/master/pvlib/singlediode_methods.py#L114) append
    ```python
    -2 * photocurrent * d2mutau / (Vbi - diode_voltage/Ns)**3
    ```
* add two new kwargs to the end, `d2mutau` and `Vbi` to the signatures, everywhere? And set `Vbi` to a constant 0.9-volts and `d2mutau` defaults to 1.4-volts

### What are good names for the new arguments?

* `intrinsic_voltage`? `builtin_voltage`? `thin_film_voltage`? some combination?
* `diffusion_distance_ratio`? `thin_film_cross_section`? `instrinsic_recombination_rate`? or should we just stick to `d2mutau`? or call it `merten_et_al_factor`?

### remaining questions:

* should users be allowed to set `Vbi`? What about tandem and triple junctions? Should we add a parameter for number of junctions, or add a note in the docstring or elsewhere for the user to adjust `Vbi` as needed?

I'll submit a PR soon, but it'd be nice to get some feedback on suggested names, any mistakes in my derivatives, or ideas about arguments and code paradigms. Thanks!

cwhanse, 2018-07-13:
`voltage_builtin` I think .  I prefer to put the quantity first. The Mertens paper references earlier work that uses the term 'built-in voltage' and that term seems to have stuck. 'intrinsic voltage' appears to be a PVsyst term.  Users should be permitted to set `voltage_builtin` via a `kwarg` but we should supply 0.9V as the default.

We could consider adding a warning that this model is only considered appropriate for amorphous silicon cells.  Validity at high forward voltage is a different issue...

mikofski, 2018-07-13:
... and CdTe, but **not** CIGS - Yes I agree, this should be emphasized. What to call `d2mutau` ? My vote is to leave it as is.

jdnewmil, 2018-07-13:
It is a composite of variables, so yes, leave it as-is.

On July 13, 2018 1:43:59 PM PDT, Mark Mikofski <notifications@github.com> wrote:
>... and CdTe, but **not** CIGS - Yes I agree, this should be
>emphasized. What to call `d2mutau` ? My vote is to leave it as is.

-- 
Sent from my phone. Please excuse my brevity.

closed 2018-08-08 by wholmgren

adriesse, 2019-08-08:
Not sure it is pythonic (or gitic) to comment on a closed issues, but it seems it would be really useful to have `d2mutau` and `NsVbi` accepted in `bishop88_mpp`, or something like that.  Has anyone got that on their radar screens?

cwhanse, 2019-08-08:
Not on my radar. I agree it would be useful.  Let's open a new issue for the enhancement request.

mikofski, 2019-08-08:
OK, but `bishop88` does already take these arguments:
https://github.com/pvlib/pvlib-python/blob/0a14b273517a082603cc157faa88a5ab0ac4cac9/pvlib/singlediode.py#L70-L72

Seems like an oversight that they're not in `bishop88_mpp`, `bishop88_i_from_v`, or ``bishop88_i_from_v`. I believe this may be an easy fix?

adriesse, 2019-08-09:
Seemed like something that fell between the cracks.  Is it really just a matter of passing the parameters through?

mikofski, 2019-08-09:
yes

adriesse, 2019-08-09:
@mikofski Care to look in my `bishop` branch?  It's all I had time for today; I think it works.  Maybe a PR next week.

PR #504 (2018-08-08): ENH: implementing pvsyst recombination loss current for CdTe and a:Si
* closes #163
* add constant VOLTAGE_BUILTIN = 0.9 (V) to singlediode_methods.py
* add arguments d2mutau=0, voltage_builtin=VOLTAGE_BUILTIN, and
cells_in_series to calculate i_recomb and v_recomb
* add check for d2mutau to calc i_recomb, v_recomb, and gradients
* add terms to current and gradients

pvlib python pull request guidelines
====================================

Thank you for your contribution to pvlib python! You may delete all of these instructions except for the list below.

You may submit a pull request with yo [... cut]
commit 2a436c60 (2018-08-08): ENH: implementing pvsyst recombination loss current for CdTe and a:Si (#504)

* implementing pvsyst recombination loss current for CdTe and a:Si

* closes #163
* add constant VOLTAGE_BUILTIN = 0.9 (V) to singlediode_methods.py
* add arguments d2mutau=0, voltage_builtin=VOLTAGE_BUILTIN, and
cells_in_series to calculate i_recomb and v_recomb
* add check for d2mutau to calc i_recomb, v_recomb, and gradients
* add terms to current and gradients

* ENH: need to set grad_i_recomb if d2mutau not given

* also set v_recomb to Inf, so that 1/v_recomb is always zero
* but precalculate grad_2i_recomb onl [... cut]