pvlib/pvlib-python #167: Overflow in clearsky.haurwitz when apparent_zenith is almost 90

cwhanse, 2016-05-12:
Haurwitz model includes this term:

np.exp(-0.059/cos_zenith)

We currently don't intercept and handle numerical overflow when cos_zenith is very close to 0.

wholmgren, 2016-05-13:
Cliff, I don't fully understand what you mean by numerical overflow in the python/numpy context since you usually will get the largest number your platform supports, a nan, or an inf. In any case, it might be platform and numpy version dependent. On my Python 3.5, 64-bit mac and my Python 2.7, 64-bit linux, both running numpy 1.10.4...

```python
In [2]: np.cos(np.deg2rad(89.99999999999999999999999999999999999))
Out[2]: 6.123233995736766e-17

In [3]: 1/np.cos(np.deg2rad(89.99999999999999999999999999999999999))
Out[3]: 16331239353195370.0

In [4]: np.exp(-0.059/np.cos(np.deg2rad(89.99999999999999999999999999999999999)))
Out[4]: 0.0
```

Same results with 90 and 90.0, and the behavior from 89.0 - 89.9 also seems ok. Can you copy/paste a few lines of input and output?

cwhanse, 2016-05-13:
Hi Will,

Perhaps it’s a runtime setting on my end.  I get warnings, not errors, when the zenith is 90 (or very close to it):

C:\Users\cwhanse\AppData\Local\Continuum\Anaconda3\lib\site-packages\pandas\computation\expressions.py:62: RuntimeWarning: invalid value encountered in power
return op(a, b)
C:\Users\cwhanse\AppData\Local\Continuum\Anaconda3\lib\site-packages\pvlib\clearsky.py:314: RuntimeWarning: overflow encountered in exp
clearsky_GHI = 1098.0 * cos_zenith * np.exp(-0.059/cos_zenith)

The code executes to end.  Matlab handles this case with the ‘Inf’ value, which made it transparent to us.

I mostly wanted to learn how to post an issue, you can close it if it’s not really an issue.

Cliff

wholmgren, 2016-05-13:
Are you using Spyder on Windows? I wonder if it's related to pydata/pandas#9950 and links therein. You might be able to eliminate the warnings by running ``np.errstate(over='ignore')`` or, more aggressively, ``np.errstate(all='ignore')``

wholmgren, 2016-05-13:
Actually, I think [np.seterr](http://docs.scipy.org/doc/numpy-1.10.1/reference/generated/numpy.seterr.html#numpy.seterr) might be the right one to use.

closed 2016-06-13 by cwhanse