pvlib/pvlib-python #177: ModelChain claims it supports module_parameters from CEC, but only uses sapm internally

jforbess, 2016-05-21:
Am I confused, or should there be a check as to whether the module params are CEC or Sandia, because you have to use different models depending on which set of parameters you're using?

wholmgren, 2016-05-22:
I certainly never meant to claim that ModelChain supported CEC modules or singlediode! I'd readily accept a PR that fixes documentation or makes it more clear. Have you seen PR #151 that refactors ModelChain and implements SAPM and SingleDiode classes? It needs to be rebased on the latest changes, but it's mostly complete. I want to get 0.3.3 out before merging that and switching development to the 0.4 series.

jforbess, 2016-05-22:
Yeah, the parameter definition is: "Module parameters as defined by the
SAPM, CEC, or other."

Ah, yes, PR #151. I thought I had seen it go through, but I only checked
Issues, not PRs. I will have to look at it when I have more time. I am
motivated to get singlediode in there, because that's what I use, but I had
to stop with the ModelChain work today when I hit this issue, because I
have a deadline, as usual.

wholmgren, 2016-05-22:
Thanks for pointing that out. I made quick update to the documentation to clarify the current status. 

I also rebased the code in #151, pushed it to a new branch, and opened a new PR from that branch.

closed 2016-05-22 by wholmgren