pvlib/pvlib-python #181: Support for Python 3

lukewitmer, 2016-05-25:
Is there a plan to support both python 2 and python 3 or to migrate fully over to python 3? https://caniusepython3.com/project/pvlib returns False but says there are no dependencies blocking transition to python 3. Any thoughts on this? Python 3 will soon (this year) be more widely supported than python 2. https://blogs.msdn.microsoft.com/pythonengineering/2016/03/08/python-3-is-winning/

wholmgren, 2016-05-25:
pvlib is fully compatible with python 2.7, 3.4, and 3.5 on all platforms:

https://travis-ci.org/pvlib/pvlib-python

I don't know anything about that website, but I think it's wrong.

I'd prefer to only support Python 3, but that's not going to happen anytime soon.

lukewitmer, 2016-05-25:
Thanks @wholmgren for that link. Good to know! All of the other packages I am using like pandas, numpy, etc return True from caniusepython3. I'm not sure if there is some annotation or something to add to the codebase to get it to return True. That website (you can also pip install caniusepython3) seems to dynamically pull the library you are interested in and checks it for support or compatibility or soemthing.
```
>>> import caniusepython3
>>> caniusepython3.check(projects=['pandas'])
No handlers could be found for logger "ciu"
True
>>> caniusepython3.check(projects=['pvlib'])
False
```

wholmgren, 2016-05-25:
After a very quick look on caniusepython3's [github page](https://github.com/brettcannon/caniusepython3), I think we need to add some classifiers to our setup.py file

https://docs.python.org/3/distutils/setupscript.html#additional-meta-data
https://pypi.python.org/pypi?%3Aaction=list_classifiers

cwhanse, 2016-05-25:
Pydap is not yet python3 compliant which is a pain when you need OpenDAP data source like NOAA NAM forecasts.  Something to keep in mind as Will is prepping a forecast module.

Thanks @wholmgren<https://github.com/wholmgren> for that link. Good to know! All of the other packages I am using like pandas, numpy, etc return True from caniusepython3. I'm not sure if there is some annotation or something to add to the codebase to get it to return True. That website (you can also pip install caniusepython3) seems to dynamically pull the library you are interested in and checks it for support or compatibility or something.

>>> import caniusepython3
>>> caniusepython3.check(projects=['pandas'])
No handlers could be found for logger "ciu"
True
>>> caniusepython3.check(projects=['pvlib'])
False

-

wholmgren, 2016-05-25:
Thanks for pointing that out, Cliff -- you're becoming a python expert! It looks like there's been some recent movement for Python 3 support in PyDAP. However, the forecast module that we wrote currently uses the Siphon package to get data from Unidata's THREDDS server, and that's all Python 3 compatible.

closed 2016-05-26 by wholmgren

wholmgren, 2016-06-16:
Following up after the new release (0.3.3) with new classifiers... caniusepython3 still reports no. I give up.

lukewitmer, 2016-06-16:
Hmm, no worries, and I think what you've done so far are steps in the right direction. The fact that it is python 3 compatible regardless of what caniusepython3 reports works for me for now. We will get it figured out eventually.

PR #183 (2016-05-26): add classifiers to setup.py
might close #181. I copy/pasted from xarray with minor modifications.