pvlib/pvlib-python #197: reorder function arguments to put variables first and parameters second

wholmgren, 2016-06-23:
I think we should reorder a handful of function arguments to put the most variable-like arguments first and most parameter-like arguments last. I'm guessing that a lot people will think that I'm just screwing with the API for no reason, but I do have a few reasons!

1. consistency is important (and underrated)
2. we can make better use of default arguments. 

Here are the functions that I know about so far:

```python
ashraeiam(b, aoi) --> ashraeiam(aoi, b=0.05)

physicaliam(K, L, n, aoi) --> physicaliam(aoi, K=4., L=0.002, n=1.526)

singlediode(module, photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth) -->
singlediode(photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth, module)

# I just noticed that the matlab sapm api is very different, so that's another issue
sapm(module, poa_direct, poa_diffuse, temp_cell, airmass_absolute, aoi) --> 
sapm(poa_direct, poa_diffuse, temp_cell, airmass_absolute, aoi, module)

snlinverter(inverter, v_dc, p_dc) --> snlinverter(v_dc, p_dc, inverter)
```

jforbess, 2016-06-24:
While this is a pain, messing with the api, I do think it makes sense, and
is worthwhile going forward.

cwhanse, 2016-06-24:
I agree, this effort makes sense in general.  A few specific tweaks:
For physicaliam, I'd order as aoi, n, K, L.  K and L can be safely set to the defaults as the function's values are almost entirely determined by n.
See my comment on #198 for the sapm.

wholmgren, 2016-06-24:
Good point, Cliff. I updated that line.

ref #200.

PR #220 (2016-07-22): improve function argument ordering in pvsystem
This PR improves some function signatures as discussed in #197.

Also addresses the input/output type issues described #201 for ashraeiam, physicaliam, and snlinverter.

Adds test_tools.py. 

I'll merge tomorrow unless there are comments.

Closes #197.
closed 2016-07-22 by wholmgren