pvlib/pvlib-python #206: Forecasted GHI is greater than clear sky GHI

willhobbs, 2016-06-30:
Using GFS as the forecast model, I am seeing GHI values as much as 13% higher than what ModelChain calculates when no irradiance data is provided (default ineichen clearsky model).

cwhanse, 2016-06-30:
I’ve seen the same thing from NAM, compared to Ineichen’s clear sky model.

wholmgren, 2016-06-30:
There are two reasons why the forecast module's irradiance data will be different from the ineichen model's: 

1. The forecast module uses the Liu Jordan model to convert a cloud cover forecast to an irradiance forecast. It's going to give different results even in clear sky conditions.
2. The forecast module fails to supply the Liu Jordan model with a time-varying top-of-atmosphere irradiance. I'll fix this later today.

@cwhanse are you referring to the raw NAM data or the forecast module's processed version of it?

cwhanse, 2016-06-30:
I’m referring to the GHI values in the NAM forecast downloaded from NOAA.

wholmgren, 2016-07-14:
@willhobbs I wrote a new cloud cover to irradiance algorithm that's based on the ineichen function and the turbidity file. It's basically this... 

```python
solpos = location.get_solarposition(cloud_cover.index)
cs = location.get_clearsky(cloud_cover.index, model='ineichen')
# offset and cloud cover in decimal units here
ghi = (offset + (1 - offset) * (1 - cloud_cover)) * ghi_clear
dni = disc(ghi, solpos['zenith'], cloud_cover.index)['dni']
dhi = ghi - dni * np.cos(np.radians(solpos['zenith']))
```

See the [updated documentation](https://forecasting.energy.arizona.edu/pvlib/pvlibfx/pvlib-python/docs/sphinx/build/html/forecasts.html#cloud-cover-and-radiation) for more.

Lot's of room for experimentation and likely improvements. I have no idea how the accuracy of the two methods ('clearsky_scaling' and 'liujordan') compares.

wholmgren, 2016-07-28:
Closed by #180.

closed 2016-07-28 by wholmgren