pvlib/pvlib-python #216: irradiadiance.extraradiation() clarification

adriesse, 2016-07-14:
The code contains the following cryptic comment:

> Notes
>     -----
>     The Spencer method contains a minus sign discrepancy between
>     equation 12 of [1]. It's unclear what the correct formula is.

I found the minus sign in reference [1], ran both versions of the formula, and found the plus version to give the smaller error with respect to pyephem.  The plus version is also given in Duffie and Beckman, so this comment can be safely removed.

wholmgren, 2016-07-14:
I'm fine with removing it. I always assumed that it was from the matlab toolbox, but I searched for it and didn't find it in the latest version. Thanks for doing the test, too. Want to make a pull request?

adriesse, 2016-07-15:
I'm not set up to do that (yet).

closed 2016-07-20 by wholmgren