pvlib/pvlib-python #221: critical error in singlediode p_mp calculation

wholmgren, 2016-07-20:
Bad news. I discovered a major error in the singlediode p_mp calculation: given a time series input, the numerical solver exits if the solution for any single time is within the error tolerance, rather than requiring that the solution for all times be close enough. The error is in [this line](https://github.com/pvlib/pvlib-python/blob/5218c9d3dd2d5a4e410140de3180b7ef00e8b1ee/pvlib/pvsystem.py#L1728). 

Here's some sample code:

```python
photocurrent = np.linspace(0, 10, 11)
resistance_shunt = 16
resistance_series = 0.094
nNsVth = 0.473
saturation_current = 1.943e-09

sd = pvsystem.singlediode(photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth)

sd['i_mp']
array([ 0.        ,  0.59453544,  1.56145498,  2.54105091,  3.52353436,
        4.50658749,  5.4890739 ,  6.47016579,  7.44909413,  8.42504244,
        9.39708533])

# change .all() to .any()

pvsystem = reload(pvlib.pvsystem)

sd_any = pvsystem.singlediode(photocurrent, saturation_current, resistance_series, resistance_shunt, nNsVth)

sd_any['i_mp']
array([ 0.        ,  0.54538398,  1.43273966,  2.36328163,  3.29255606,
        4.23101358,  5.16177031,  6.09368251,  7.02197553,  7.96846051,
        8.88220557])

sd['i_mp']/sd_any['i_mp']
array([        nan,  1.09012266,  1.0898386 ,  1.07522137,  1.07015167,
        1.06513189,  1.06340917,  1.06178256,  1.06082599,  1.05729864,
        1.05796756])
```

So the error here is on the order of 5 to 10%, but the results will depend on your exact input.

The blue dots on the IV curve below show the p_mp points in the old code. The black dots show the p_mp points in fixed code. These IV curves are generated using a fix for #83 that I'll make a PR for soon. They're also what led me to discovering the problem.

[image] 

On a related note, someone should put more effort into verifying that the matlab and python libraries give the same results. Not me.

jforbess, 2016-07-21:
Wow. Good catch.

jforbess, 2016-07-21:
Also, will this fix trigger a new release 0.3.4? Seems pretty important.

wholmgren, 2016-07-21:
I want to release 0.4 next week, but I am sympathetic to the fact that people may not be happy about dealing with a lot of API changes just so that they can get this fix. So, I figured that I'd wait to see if anyone requested it.

jforbess, 2016-07-21:
Ok. I have a project I'm dealing with this week, but I can make the
individual fix myself. I didn't realize 0.4.0 was so close. I can accept
the API changes as well as the fix, so I'll let others request if they need
it.

cwhanse, 2016-07-21:
Will,

I'll take this on - test case for single diode functions.

Can you point me to an example of a test implemented as you like it?

Cliff

[ivcurves]<https://cloud.githubusercontent.com/assets/4383303/17006598/cedaa96c-4e96-11e6-9e9a-64ed1377536c.png>

closed 2016-07-22 by wholmgren

wholmgren, 2016-07-22:
Thanks, Cliff. 

You could base your test on the example above, but you probably don't need as many points. 

For slightly more realistic input, you could base your test on something similar to the test suite's [test_singlediode_series_ivcurve](https://github.com/wholmgren/pvlib-python/blob/a64d43280b9760ae30d3e2dc27587695960320da/pvlib/test/test_pvsystem.py#L455) function. 

```python
def test_singlediode_series_ivcurve(cec_module_params):
    times = pd.DatetimeIndex(start='2015-06-01', periods=3, freq='6H')
    poa_data = pd.Series([0, 400, 800], index=times)
    IL, I0, Rs, Rsh, nNsVth = pvsystem.calcparams_desoto(
                                         poa_data,
                                         temp_cell=25,
                                         alpha_isc=cec_module_params['alpha_sc'],
                                         module_parameters=cec_module_params,
                                         EgRef=1.121,
                                         dEgdT=-0.0002677)

    out = pvsystem.singlediode(IL, I0, Rs, Rsh, nNsVth, ivcurve_pnts=3)

    expected = OrderedDict([('i_sc', array([        nan,  3.01054475,  6.00675648])),
             ('v_oc', array([         nan,   9.96886962,  10.29530483])),
             ('i_mp', array([        nan,  2.65191983,  5.28594672])),
             ('v_mp', array([        nan,  8.33392491,  8.4159707 ])),
             ('p_mp', array([         nan,  22.10090078,  44.48637274])),
             ('i_x', array([        nan,  2.88414114,  5.74622046])),
             ('i_xx', array([        nan,  2.04340914,  3.90007956])),
             ('v',
              array([[         nan,          nan,          nan],
       [  0.        ,   4.98443481,   9.96886962],
       [  0.        ,   5.14765242,  10.29530483]])),
             ('i',
              array([[             nan,              nan,              nan],
       [  3.01079860e+00,   2.88414114e+00,   3.10862447e-14],
       [  6.00726296e+00,   5.74622046e+00,   0.00000000e+00]]))])

    for k, v in out.items():
        assert_allclose(expected[k], v, atol=1e-2)
```

The function is called with the 'Example_Module' parameters. Note that the numbers above are generated using the corrected solver.

PR #222 (2016-07-22): Fix major error in singlediode p_mp calculation for array/Series input
Fixes #221. 

I'll merge tomorrow.