pvlib/pvlib-python #237: Localized time index not working when using 'nrel_c' method

anomam, 2016-08-12:
Hi,

I'm not sure if that was already covered in an other issue (couldn't find it) or intentional in the code, but when using the function ``get_solarposition()`` with ``method='nrel_c''', it doesn't seem to matter whether the time index is localized or not.
The function will assume that the time index is at 'UTC' no matter what. Which goes against what was described in http://pvlib-python.readthedocs.io/en/latest/timetimezones.html.

So I ended up converting my timezone to 'UTC' before calculation, and then back to local after the calculation

Example:
``pvlib_loc = location.Location(latitude, longitude, tz='Etc/GMT+8')
localized_index = index.tz_localize('Etc/GMT+8')
utc_index = localized_index.tz_convert('UTC')
solarAngles = pvlib_loc.get_solarposition(utc_index, method='nrel_c')
solarAngles.index = localized_index``

wholmgren, 2016-08-17:
I'd welcome a pull request if you want to fix it or at least document the bug. Personally, I'd vote for removing the nrel_c option all together, but that's probably a separate issue. Is there a reason you're using nrel_c instead of the nrel_numpy or nrel_numba options?

mikofski, 2016-11-04:
At the very least this should be changed in the docstring I think.

PR #333 (2017-06-03): Allowing spa_c() to use localized time index 0.4.5
continuation of #301. closes #237.
closed 2017-06-03 by wholmgren