pvlib/pvlib-python #242: lines in snlinverter table in docs doesn't wrap

mikofski, 2016-09-12:
the table in the docs for [`pvlib.pvsystem.snlinverter`](http://pvlib-python.readthedocs.io/en/latest/modules.html#pvlib.pvsystem.snlinverter) doesn't wrap, causing the text before and after the table and in the table itself to extend off the page.

see [tables in sphinx/docutils docs](http://www.sphinx-doc.org/en/stable/rest.html#tables) and [this SO Q/A](http://stackoverflow.com/questions/16570087/multiple-lines-of-text-in-single-cell-of-simple-table),

>simple table style does not support wrapping blocks. Use the grid style instead

wholmgren, 2016-09-12:
Thanks for pointing that out. You're already working on the fix, right???

dacoex, 2016-09-13:
I recommend using a CSV-table.
I found it easier to handle & to maintain.

Here, it does not wrap either but use a scrollbar:
http://pvlib-python.readthedocs.io/en/latest/variables_style_rules.html

Back when I wrote the page, I faced a similar issue.
It was due to the CSS of RTD.
See also:
http://stackoverflow.com/questions/34194728/csv-table-formatting-in-python-docstrings-sphinx-multiple-lines-in-one-cell

--> So check also against a "blank" sphinx project.

wholmgren, 2016-09-13:
I think we need to make the table inline if we want the function documentation to be properly rendered using help, ?, or ??

wholmgren, 2016-09-14:
@mikofski I'd prefer that we close this issue, but am ok with leaving it open if you think you'll return to it at some point.

mikofski, 2016-09-14:
@wholmgren PR #243 fixes this issue enough for me, so I also prefer to close this issue. Thanks!

closed 2016-09-14 by mikofski