pvlib/pvlib-python #265: fix linke turbidity time interpolation

mikofski, 2016-11-16:
As indicated in [in `clearsky.lookup_linke_turbidity` on L212-224](../blob/master/pvlib/clearsky.py#L212-L224)

> This is approximate and could be made more accurate.

```python
>>> import numpy as np
>>> np.linspace(-15, 380, num=14)
array([ -15.        ,   15.38461538,   45.76923077,   76.15384615,
        106.53846154,  136.92307692,  167.30769231,  197.69230769,
        228.07692308,  258.46153846,  288.84615385,  319.23076923,
        349.61538462,  380.        ])
```

Another method would be to use the calendar month days:

```python
>>> import calendar
>>> import numpy as np
>>> mdays = calendar.mdays[1:]
>>> np.concatenate([[-calendar.mdays[1] / 2.0],
                    np.cumsum(mdays) - np.array(mdays) / 2.,
                    [365 + calendar.mdays[-1] / 2.0]])
array([ -15.5,   15.5,   45. ,   74.5,  105. ,  135.5,  166. ,  196.5,
        227.5,  258. ,  288.5,  319. ,  349.5,  380.5])
```

This still doesn't account for leap years, but we can use `calendar.isleap(<year>)`:
```python
import calendar
import numpy as np

def calendar_month_middles(year):
    mdays = np.array(calendar.mdays[1:])  # copy
    if calendar.isleap(year):
        mdays[1] = mdays[1] + 1
    return np.concatenate([[-calendar.mdays[1] / 2.0],
                    np.cumsum(mdays) - np.array(mdays) / 2.,
                    [365 + calendar.mdays[-1] / 2.0]])
```

now you can see the difference in leap year

```python
>>> calendar_month_middles(2016)  # a leap year
array([ -15.5,   15.5,   45.5,   75.5,  106. ,  136.5,  167. ,  197.5,
        228.5,  259. ,  289.5,  320. ,  350.5,  380.5])
>>> calendar_month_middles(2015)  # NOT a leap year
array([ -15.5,   15.5,   45. ,   74.5,  105. ,  135.5,  166. ,  196.5,
        227.5,  258. ,  288.5,  319. ,  349.5,  380.5])
```

wholmgren, 2016-11-16:
I'm guessing that the biggest practical impact of this will be that people see an example of the ``calendar`` module, but sounds good to me!

mikofski, 2016-11-16:
Do you want me to add this to pull request #264 ? 

Also I forgot to change 365 to 366 in `calendar_month_middles()`

wholmgren, 2016-11-16:
Sure. It might make sense to do it all at once if it's going to change the
unit test expected values.

commit 300d1ab2 (2016-11-18): Linke turbidity factor fixes (#264)

* fix #263 and #262

* set minimum index as 0 not 1
* adjust _linearly_scale so that is increments are 5 minutes or 1/12 of an
 arcdegree from the centers of neighboring indices
* this fixes the out of bounds error for latitudes or longitudes at the
 limits of the range
* move _linearly_scale to be right after lookup_linke_turbidity
* add link to pdf on AOD and LT from MeteoTest
* raise IndexError if outputmatrix is more than half an index outside of
 the limits

* unset debug logging

* remove logging and use pytest with raises context

* all test passing  [... cut]
closed 2016-11-18 by wholmgren