pvlib/pvlib-python #267: Tutorial: pvsystem.ipynb

cdeline, 2016-11-18:
I'm getting some errors when running pvsystem.ipynb in the PVLib tutorials directory.  Specifically code block 17 should be replaced with 
`pacs = pvsystem.snlinverter(vdcs, pdcs, inverters['ABB__MICRO_0_25_I_OUTD_US_208_208V__CEC_2014_'])`

Also, code block 22 is throwing an error at `irrad_data = clearsky.ineichen(times_loc, tus.latitude, tus.longitude)`.  It looks like this code example is passing local datetime into clearsky.ineichen, rather than apparent_zenith angle in degrees.

I'm using the latest pvlib 0.4.1 python 2.7.

Chris Deline

wholmgren, 2016-11-18:
Thanks for pointing that out and sorry for the confusion. We made some API changes in 0.4 and neglected to update many of the jupyter/ipython tutorials. The online documentation should be up to date: http://pvlib-python.readthedocs.io 

Your second error can be fixed with: ``irrad_data = tus.get_clearsky(times_loc)``

wholmgren, 2016-11-18:
This one should work better:

http://nbviewer.jupyter.org/github/wholmgren/pvlib-python/blob/tutorials041/docs/tutorials/pvsystem.ipynb

I'll add this to a future release.

closed 2016-12-07 by wholmgren

cdeline, 2016-12-07:
Thanks guys!

Chris Deline
National Renewable Energy Laboratory
303-384-6359

Closed #267<https://github.com/pvlib/pvlib-python/issues/267> via #281<https://github.com/pvlib/pvlib-python/pull/281>.

PR #281 (2016-12-07): update pvsystem and tracking tutorials for pvlib 0.4 api changes
closes #267. 

Also adding @gperonato to the whatsnew file for #275.