pvlib/pvlib-python #271: io module: add macc-rad reader

dacoex, 2016-12-01:
**Upstream data source**
http://www.soda-pro.com/web-services/radiation/cams-radiation-service

**Git repository with examples**
https://github.com/dacoex/pvlib_data

**Related**
Issues: #261 & #29
PR: #270 

**Previous comments**
taken from:
https://github.com/pvlib/pvlib-python/pull/270#issuecomment-264102908

> The header section of the maccrad file looks like it could be of variable length, so you probably don't want to hardcode the 40. The leading # can be used to let read_csv skip the variable header section, but unfortunately the column names would be lost that way too. Is the number of columns always the same? In that case I would suggest using header=None and providing a standard name for each column.
> 
> Your dateparser looks like it returns text. Does this produce a datetime index for the dataframe?
> 
> I the time zone for the data in the file? "# Time reference: Universal time (UT)" Maybe the dataframe could be localized with this?

dacoex, 2016-12-01:
>The file maccrad format doesn't look like it was designed for ease of parsing.

It says that there is some standard befind:

_ Encoding partly from D2.8.III.13-14 INSPIRE Data Specification on Atmospheric Conditions and Meteorological Geographical Features – Technical Guidelines, 2013-12-10; and CF (Climate and Forecast) metadata (2013-11-11) CF Standard Names registry of ObservablePropertyValue_

>Is the number of columns always the same?

Yes.

>Your dateparser looks like it returns text. Does this produce a datetime index for the dataframe?

Yes.

>I the time zone for the data in the file? "# Time reference: Universal time (UT)" Maybe the dataframe >could be localized with this?

Thanks, I'd forgot but now implemented.
The functions also add metadata to the dataframe (df.df_name = "MACC-RAD v2.7 Solar Irradiation") and set a default location name (loc.name = "MACC-RAD v2.7 Solar Irradiation @ lat (deg. N), lon (deg. E): 44.083, 5.059")

If the file structure & functions settle down I will add 
- io module documentation
- whatsnew

wholmgren, 2016-12-01:
Please add the discussion above to the new PR that I requested in #271, and then close the issue.

dacoex, 2016-12-02:
closed in favour of #274

closed 2016-12-02 by dacoex